ros-infrastructure/rosdep #851: rosdep update error

jqhuang112358, 2022-01-25:
hello，friends
error happend like following:
ERROR: Rosdep experienced an error: mapping values are not allowed here
  in "<unicode string>", line 60, column 12:
          debug: false

jqhuang112358, 2022-01-25:
More tips below:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 146, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 441, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 450, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 659, in command_update
    update_sources_list(success_handler=update_success_handler,
  File "/usr/lib/python3/dist-packages/rosdep2/sources_list.py", line 499, in update_sources_list
    distribution_names = get_index().distributions.keys()
  File "/usr/lib/python3/dist-packages/rosdep2/rosdistrohelper.py", line 69, in get_index
    _RDCache.index = rosdistro.get_index(_RDCache.index_url)
  File "/usr/lib/python3/dist-packages/rosdistro/__init__.py", line 105, in get_index
    data = yaml.safe_load(yaml_str)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 162, in safe_load
    return load(stream, SafeLoader)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 114, in load
    return loader.get_single_data()
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 49, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 58, in compose_document
    self.get_event()
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 118, in get_event
    self.current_event = self.state()
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 193, in parse_document_end
    token = self.peek_token()
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 129, in peek_token
    self.fetch_more_tokens()
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 223, in fetch_more_tokens
    return self.fetch_value()
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 577, in fetch_value
    raise ScannerError(None, None,
yaml.scanner.ScannerError: mapping values are not allowed here
  in "<unicode string>", line 60, column 12:
          debug: false,
               ^

nuclearsandwich, 2022-01-25:
In order to investigate this issue please provide the following information

* Operating system
* Operating system version
* rosdep version
* Output from `cat /etc/ros/rosdep/sources.list.d/20-default.list`
* Any additional rosdep sources that you've configured.

tfoote, 2022-03-08:
Closing due to no response.

closed 2022-03-08 by tfoote