ros-infrastructure/rosdep #852: Error in rosdep install (Docker) {Help Needed!}

dsriaditya999, 2022-01-25:
I faced the following error while building a Docker File

````
Processing triggers for libc-bin (2.31-0ubuntu9.2) ...
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package ros-rolling-ros-ign-bridge
Reading package lists...
Building dependency tree...
Reading state information...
The following NEW packages will be installed:
  ros-rolling-ros-ign-gazebo
0 upgraded, 1 newly installed, 0 to remove and 5 not upgraded.
Need to get 135 kB of archives.
After this operation, 593 kB of additional disk space will be used.
Get:1 http://packages.ros.org/ros2/ubuntu focal/main amd64 ros-rolling-ros-ign-gazebo amd64 0.244.1-1focal.20220105.183030 [135 kB]
Fetched 135 kB in 2s (78.1 kB/s)
Selecting previously unselected package ros-rolling-ros-ign-gazebo.
(Reading database ... 138697 files and directories currently installed.)
Preparing to unpack .../ros-rolling-ros-ign-gazebo_0.244.1-1focal.20220105.183030_amd64.deb ...
Unpacking ros-rolling-ros-ign-gazebo (0.244.1-1focal.20220105.183030) ...
Setting up ros-rolling-ros-ign-gazebo (0.244.1-1focal.20220105.183030) ...
Reading package lists...
Building dependency tree...
Reading state information...
ros-rolling-moveit-ros-move-group is already the newest version (2.3.2-1focal.20220105.205454).
ros-rolling-moveit-ros-move-group set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
The following NEW packages will be installed:
  ros-rolling-moveit-servo
0 upgraded, 1 newly installed, 0 to remove and 5 not upgraded.
Need to get 671 kB of archives.
After this operation, 4743 kB of additional disk space will be used.
Get:1 http://packages.ros.org/ros2/ubuntu focal/main amd64 ros-rolling-moveit-servo amd64 2.3.2-1focal.20220105.210716 [671 kB]
Fetched 671 kB in 3s (257 kB/s)
Selecting previously unselected package ros-rolling-moveit-servo.
(Reading database ... 138721 files and directories currently installed.)
Preparing to unpack .../ros-rolling-moveit-servo_2.3.2-1focal.20220105.210716_amd64.deb ...
Unpacking ros-rolling-moveit-servo (2.3.2-1focal.20220105.210716) ...
Setting up ros-rolling-moveit-servo (2.3.2-1focal.20220105.210716) ...
Processing triggers for libc-bin (2.31-0ubuntu9.2) ...
Continuing to install resolvable dependencies...
executing command [apt-get install -y ros-rolling-moveit-ros-planning-interface]
executing command [apt-get install -y ros-rolling-ros-ign-bridge]
executing command [apt-get install -y ros-rolling-ros-ign-gazebo]
executing command [apt-get install -y ros-rolling-moveit-ros-move-group]
executing command [apt-get install -y ros-rolling-moveit-servo]
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ign_moveit2_examples: Cannot locate rosdep definition for [pymoveit2]
panda_moveit_config: Cannot locate rosdep definition for [ign_ros2_control]
ERROR: the following rosdeps failed to install
  apt: command [apt-get install -y ros-rolling-ros-ign-bridge] failed
  apt: Failed to detect successful installation of [ros-rolling-ros-ign-bridge]
The command '/bin/bash -c mkdir -p drl_grasping/src &&     cd drl_grasping/src &&     git clone https://github.com/AndrejOrsula/drl_grasping.git --recursive --depth 1 -b ${DRL_GRASPING_GIT_BRANCH} &&     vcs import < drl_grasping/drl_grasping.repos &&     cd .. &&     apt update &&     source /opt/ros/${ROS2_DISTRO}/setup.bash &&     rosdep update &&     rosdep install -r --from-paths . -yi --rosdistro ${ROS2_DISTRO} &&     rm -rf /var/lib/apt/lists/* &&     source /opt/ros/${ROS2_DISTRO}/setup.bash &&     colcon build --merge-install --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release &&     source install/local_setup.bash' returned a non-zero code: 1
```

dsriaditya999, 2022-01-27:
@nuclearsandwich can you please look into this? Any suggestions would be really helpful.

nuclearsandwich, 2022-01-31:
@dsriaditya999 please ask questions on [ROS Answers](https://answers.ros.org) per the [ROS support policy](https://www.ros.org/blog/getting-started/#support). The error installing this package is caused by the fact that that package is [not currently building on ROS Rolling](https://build.ros2.org/job/Rbin_uF64__ros_ign_bridge__ubuntu_focal_amd64__binary/).

That build failure will likely be addressed in a forthcoming release of ros_ign.
A fix has been merged with https://github.com/ignitionrobotics/ros_ign/pull/212

closed 2022-01-31 by nuclearsandwich