ros-infrastructure/rosdep #857: Could not detect OS (raspbian)

christianrauch, 2022-02-13:
I am running rosdep on Raspberry Pi OS and got the following error:
```
ERROR: Rosdep experienced an error: Could not detect OS, tried ['zorin', 'windows', 'nixos', 'clearlinux', 'ubuntu', 'slackware', 'rocky', 'rhel', 'qnx', 'pop', 'osx', 'sailfishos', 'tizen', 'oracle', 'opensuse', 'opensuse', 'opensuse', 'opensuse', 'opensuse', 'openembedded', 'neon', 'mx', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora', 'elementary', 'elementary', 'debian', 'cygwin', 'euleros', 'centos', 'manjaro', 'buildroot', 'arch', 'amazon', 'alpine', 'almalinux']
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.21.0

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 146, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 441, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 450, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 659, in command_update
    update_sources_list(success_handler=update_success_handler,
  File "/usr/lib/python3/dist-packages/rosdep2/sources_list.py", line 516, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python3/dist-packages/rosdep2/gbpdistro_support.py", line 150, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/usr/lib/python3/dist-packages/rosdep2/__init__.py", line 91, in create_default_installer_context
    m.register_platforms(context)
  File "/usr/lib/python3/dist-packages/rosdep2/platforms/debian.py", line 65, in register_platforms
    register_elementary(context)
  File "/usr/lib/python3/dist-packages/rosdep2/platforms/debian.py", line 95, in register_elementary
    (os_name, os_version) = context.get_os_name_and_version()
  File "/usr/lib/python3/dist-packages/rosdep2/installers.py", line 115, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/usr/lib/python3/dist-packages/rospkg/os_detect.py", line 610, in get_name
    self.detect_os()
  File "/usr/lib/python3/dist-packages/rospkg/os_detect.py", line 580, in detect_os
    raise OsNotDetected("Could not detect OS, tried %s" % attempted)
rospkg.os_detect.OsNotDetected: Could not detect OS, tried ['zorin', 'windows', 'nixos', 'clearlinux', 'ubuntu', 'slackware', 'rocky', 'rhel', 'qnx', 'pop', 'osx', 'sailfishos', 'tizen', 'oracle', 'opensuse', 'opensuse', 'opensuse', 'opensuse', 'opensuse', 'openembedded', 'neon', 'mx', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora', 'elementary', 'elementary', 'debian', 'cygwin', 'euleros', 'centos', 'manjaro', 'buildroot', 'arch', 'amazon', 'alpine', 'almalinux']
```

The OS is:
```
$ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

The script may look for `ID` (`raspbian`, which is not in the set above) where alternatively, it should also look for `ID_LIKE` (`debian`, which is in the set above).

igor-markovic, 2022-02-22:
I ran into the same issue, this seems to be a bug in rospkg. I think it's related to this ticket https://github.com/ros-infrastructure/rospkg/issues/241

auser, 2022-03-27:
Did you ever get this set? I'm running into this issue too

christianrauch, 2022-03-27:
You can override the OS with `--os`:
```sh
rosdep install --from-paths src --ignore-src -y --os=debian:bullseye
```

denesh-globotix, 2022-04-11:
Just got the above stack trace as well. My OS version is: 
```
cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```
This was part of step 5 of [this tutorial](https://varhowto.com/install-ros-noetic-raspberry-pi-4/#rosdep_init) trying to do ```rosdep update```

nuclearsandwich, 2022-05-23:
https://github.com/ros-infrastructure/rospkg/pull/244 detects Raspbian distinct from Debian and https://github.com/ros-infrastructure/rosdep/pull/867 has rosdep treat Raspbian as Debian.

closed 2022-05-23 by nuclearsandwich