ros-infrastructure/rosdep #868: rosdep update error for no network problem

Horizonzhou, 2022-04-25:
reading in sources list data from /etc/ros/rosdep/sources.list.d
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml]:
	nonnumeric port: 'port' (https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml)
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml]:
	nonnumeric port: 'port' (https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml)
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml]:
	nonnumeric port: 'port' (https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml)
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml]:
	nonnumeric port: 'port' (https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml)
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml]:
	Failed to download target platform data for gbpdistro:
	nonnumeric port: 'port'
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml

ERROR: Rosdep experienced an error: nonnumeric port: 'port'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.21.0

cottsay, 2022-05-09:
What platform is this? I can't reproduce the problem.

ZenithalHourlyRate, 2022-05-15:
This may be resolved by https://github.com/ros-infrastructure/rosdep/pull/839#issuecomment-1126991271