ros-infrastructure/rosdep #869: Rosdep Update error: module 'os' has no attribute 'geteuid'

adriennewinter, 2022-04-27:
The below is the error that comes up while running on Windows cmd prompt.

ERROR: Rosdep experienced an error: module 'os' has no attribute 'geteuid'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.21.0

Traceback (most recent call last):
  File "c:\python38\lib\site-packages\rosdep2\main.py", line 146, in rosdep_main
    exit_code = _rosdep_main(args)
  File "c:\python38\lib\site-packages\rosdep2\main.py", line 441, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "c:\python38\lib\site-packages\rosdep2\main.py", line 450, in _no_args_handler
    return command_handlers[command](options)
  File "c:\python38\lib\site-packages\rosdep2\main.py", line 659, in command_update
    update_sources_list(success_handler=update_success_handler,
  File "c:\python38\lib\site-packages\rosdep2\sources_list.py", line 516, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "c:\python38\lib\site-packages\rosdep2\gbpdistro_support.py", line 150, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "c:\python38\lib\site-packages\rosdep2\__init__.py", line 85, in create_default_installer_context
    m.register_installers(context)
  File "c:\python38\lib\site-packages\rosdep2\platforms\source.py", line 53, in register_installers
    context.set_installer(SOURCE_INSTALLER, SourceInstaller())
  File "c:\python38\lib\site-packages\rosdep2\platforms\source.py", line 201, in __init__
    super(SourceInstaller, self).__init__(source_detect, supports_depends=True)
  File "c:\python38\lib\site-packages\rosdep2\installers.py", line 328, in __init__
    self.sudo_command = 'sudo -H' if os.geteuid() != 0 else ''
AttributeError: module 'os' has no attribute 'geteuid'

closed 2022-04-28 by adriennewinter

Argzero, 2022-05-25:
I was able to get update to run by replacing that line with a ''.
Note: This doesn't account for the actual behavior suggested but sudo isn't a thing in the Windows PowerShell console generally. It doesn't match the workflow. Usually admin permission is obtained by program by how it is started. I would recommend checking this in a different way if needed for another reason but there is no need to append sudo -H on windows.

cottsay, 2022-05-25:
@Argzero, this should be addressed by #811, which hasn't been released yet.