ros-infrastructure/rosdep #872: requested bug report

dcanthony68, 2022-05-20:
lease go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.21.0

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 146, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 441, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 450, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 659, in command_update
    update_sources_list(success_handler=update_success_handler,
  File "/usr/lib/python3/dist-packages/rosdep2/sources_list.py", line 516, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python3/dist-packages/rosdep2/gbpdistro_support.py", line 150, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/usr/lib/python3/dist-packages/rosdep2/__init__.py", line 91, in create_default_installer_context
    m.register_platforms(context)
  File "/usr/lib/python3/dist-packages/rosdep2/platforms/debian.py", line 65, in register_platforms
    register_elementary(context)
  File "/usr/lib/python3/dist-packages/rosdep2/platforms/debian.py", line 95, in register_elementary
    (os_name, os_version) = context.get_os_name_and_version()
  File "/usr/lib/python3/dist-packages/rosdep2/installers.py", line 115, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/usr/lib/python3/dist-packages/rospkg/os_detect.py", line 634, in get_name
    self.detect_os()
  File "/usr/lib/python3/dist-packages/rospkg/os_detect.py", line 604, in detect_os
    raise OsNotDetected("Could not detect OS, tried %s" % attempted)
rospkg.os_detect.OsNotDetected: Could not detect OS, tried ['zorin', 'windows', 'nixos', 'clearlinux', 'ubuntu', 'slackware', 'rocky', 'rhel', 'qnx', 'pop', 'osx', 'sailfishos', 'tizen', 'conda', 'oracle', 'opensuse', 'opensuse', 'opensuse', 'opensuse', 'opensuse', 'openembedded', 'neon', 'mx', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora', 'elementary', 'elementary', 'debian', 'cygwin', 'euleros', 'centos', 'manjaro', 'buildroot', 'arch', 'amazon', 'alpine', 'almalinux']

nuclearsandwich, 2022-05-23:
@dcanthony68 what operating system are you running on. The error message indicates that your OS was not detected.

dcanthony68, 2022-05-23:
Raspberry Pi OS.

David Anthony

[aptiv]<http://www.aptiv.com/>

***@***.******@***.***>
Office: +1 765.867.4248  | Mobile: +1 765.398.7679

nuclearsandwich, 2022-05-23:
> Raspberry Pi OS.

This is likely due to the fact that [Raspbian](https://www.raspbian.org/) now identifies itself as distinct from Debian. You will have to override the detected OS when using rosdep. For example

```
rosdep install --os debian:bullseye ...
```

nuclearsandwich, 2022-05-23:
Future releases of rosdep will handle this aliasing automatically https://github.com/ros-infrastructure/rosdep/pull/867

closed 2022-05-23 by nuclearsandwich