ros-infrastructure/rosdep #873: Error output on versioned pip package

bbk-riact, 2022-05-27:
Package description:
```
ur-rtde-pip:
  ubuntu:
    pip:
      packages: [ur-rtde==1.5.0]
```

Terminal output:
```
executing command [pip3 install -U -q ur-rtde==1.5.0]
ERROR: the following rosdeps failed to install
  pip: Failed to detect successful installation of [ur-rtde==1.5.0]
```

The installation is actually successful.

Am I doing the definitions wrong, or is this case unsupported?

cottsay, 2022-05-27:
This is not supported right now. None of the other package managers support version constraints either.

nuclearsandwich, 2022-05-27:
I would say that specifying the version in the rosdep definition like that is unsupported.

Installation is working because rosdep is shelling out to pip and trying to install a package name `ur-rtde==1.5.0` which works because that's a valid thing to put through the pip CLI.

However, when rosdep tries to verify the installation it's checking for a package _named_ `ur-rtde==1.5.0` verbatim. And, of course, the package name is just `ur-rtde` so it doesn't find the package name it expects and raises the error.

If I recall correctly, rosdep has minimal to no handling for version constraints even though they can be expressed in package.xml dependencies. Part of the reason for this lack of support is that it isn't easy to do in a way that is portable between the various installers that rosdep supports.

When coupled with the fact that ROS itself follows a distribution model where there is just one version of any particular package available in that distribution, version constraints are not particularly broadly utilized.

One of the challenges with implementing version constraints in pip-specifically is that rosdep ideally _should not_ make changes to your system beyond installing the packages requested and since pip offers multiple versions of packages, it's possible for different packages to request conflicting versions of a dependency or for a rosdep install to remove a different version of a package required by something else on your system. That problem of conflicting / overriding versions goes away when using a distribution model where there is exactly one version of a package in the distribution at all times, which is one of the reasons that official packages on the build farm cannot use dependencies from pip at all.

At the moment I would say that if your package has version constraints on pip packages, you'll probably need to indicate those out-of-band and provide instructions for installing the requisite versions without rosdep.

bbk-riact, 2022-07-14:
@nuclearsandwich that seems like a difficult issue with no obvious solution. I'll close the issue.

closed 2022-07-14 by bbk-riact