ros-infrastructure/rosdep #875: Run without root

aboseley, 2022-05-29:
This would then be similar to other non-distro installers like rustup or ghcup.

The source list should be stored in a non-root owned folder, and then rather than installing directly the script would output the necessary commands the uses should then run as root.

cottsay, 2022-05-29:
This is already possible without any changes to `rosdep` or other ROS infrastructure packages using an environment variable:

```bash
$ ROSDEP_SOURCE_PATH=~/.ros/rosdep/sources.list.d; export ROSDEP_SOURCE_PATH
$ mkdir -p $ROSDEP_SOURCE_PATH
$ rosdep init
Wrote ~/.ros/rosdep/sources.list.d/20-default.list
Recommended: please run

	rosdep update

$ rosdep update
reading in sources list data from ~/.ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Skip end-of-life distro "dashing"
Skip end-of-life distro "eloquent"
Add distro "foxy"
Add distro "galactic"
Skip end-of-life distro "groovy"
Add distro "humble"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Skip end-of-life distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
Add distro "rolling"
updated cache in ~/.ros/rosdep/sources.cache
```

aboseley, 2022-05-30:
awesome,  thankyou!