ros-infrastructure/rosdep #878: Duplicate package name 'clpe' exists in repository 'clpe_sdk' as well as in repository 'clpe'

hhemken, 2022-06-10:
On raspberry pi 4 with Ubuntu 22.04 following the instructions at:
https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html

ERROR: Rosdep experienced an error: Duplicate package name 'clpe' exists in repository 'clpe_sdk' as well as in repository 'clpe'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.21.0

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 146, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 441, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 450, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 659, in command_update
    update_sources_list(success_handler=update_success_handler,
  File "/usr/lib/python3/dist-packages/rosdep2/sources_list.py", line 516, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python3/dist-packages/rosdep2/gbpdistro_support.py", line 149, in get_gbprepo_as_rosdep_data
    distro_file = get_release_file(gbpdistro)
  File "/usr/lib/python3/dist-packages/rosdep2/rosdistrohelper.py", line 76, in get_release_file
    dist_file = rosdistro.get_distribution_file(get_index(), distro)
  File "/usr/lib/python3/dist-packages/rosdistro/__init__.py", line 120, in get_distribution_file
    return create_distribution_file(dist_name, data)
  File "/usr/lib/python3/dist-packages/rosdistro/distribution_file.py", line 134, in create_distribution_file
    dist_file = DistributionFile(dist_name, d)
  File "/usr/lib/python3/dist-packages/rosdistro/distribution_file.py", line 63, in __init__
    self._add_package(pkg_name, repo)
  File "/usr/lib/python3/dist-packages/rosdistro/distribution_file.py", line 112, in _add_package
    assert pkg_name not in self.release_packages, "Duplicate package name '%s' exists in repository '%s' as well as in repository '%s'" % (pkg_name, repo.name, self.release_packages[pkg_name].repository_name)
AssertionError: Duplicate package name 'clpe' exists in repository 'clpe_sdk' as well as in repository 'clpe'

zwilcox, 2022-06-10:
I'm also experiencing this issue.

Here's my call stack:
```
rosdep update --include-eol-distros

ERROR: Rosdep experienced an error: Duplicate package name 'clpe' exists in repository 'clpe_sdk' as well as in repository 'clpe'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.21.0

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 146, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 441, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 450, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 662, in command_update
    ros_distro=options.ros_distro)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 516, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python2.7/dist-packages/rosdep2/gbpdistro_support.py", line 149, in get_gbprepo_as_rosdep_data
    distro_file = get_release_file(gbpdistro)
  File "/usr/lib/python2.7/dist-packages/rosdep2/rosdistrohelper.py", line 76, in get_release_file
    dist_file = rosdistro.get_distribution_file(get_index(), distro)
  File "/usr/lib/python2.7/dist-packages/rosdistro/__init__.py", line 120, in get_distribution_file
    return create_distribution_file(dist_name, data)
  File "/usr/lib/python2.7/dist-packages/rosdistro/distribution_file.py", line 134, in create_distribution_file
    dist_file = DistributionFile(dist_name, d)
  File "/usr/lib/python2.7/dist-packages/rosdistro/distribution_file.py", line 63, in __init__
    self._add_package(pkg_name, repo)
  File "/usr/lib/python2.7/dist-packages/rosdistro/distribution_file.py", line 112, in _add_package
    assert pkg_name not in self.release_packages, "Duplicate package name '%s' exists in repository '%s' as well as in repository '%s'" % (pkg_name, repo.name, self.release_packages[pkg_name].repository_name)
AssertionError: Duplicate package name 'clpe' exists in repository 'clpe_sdk' as well as in repository 'clpe'

reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Add distro "ardent"
Add distro "bouncy"
Add distro "crystal"
Add distro "dashing"
Add distro "eloquent"
Add distro "foxy"
Traceback (most recent call last):
```

xydesa, 2022-06-10:
I'm also getting this error on Linux Mint 20.3 (ubuntu):
```
rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Skip end-of-life distro "dashing"
Skip end-of-life distro "eloquent"
Add distro "foxy"

ERROR: Rosdep experienced an error: Duplicate package name 'clpe' exists in repository 'clpe_sdk' as well as in repository 'clpe'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.21.0

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 146, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 441, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 450, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 659, in command_update
    update_sources_list(success_handler=update_success_handler,
  File "/usr/lib/python3/dist-packages/rosdep2/sources_list.py", line 516, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python3/dist-packages/rosdep2/gbpdistro_support.py", line 149, in get_gbprepo_as_rosdep_data
    distro_file = get_release_file(gbpdistro)
  File "/usr/lib/python3/dist-packages/rosdep2/rosdistrohelper.py", line 76, in get_release_file
    dist_file = rosdistro.get_distribution_file(get_index(), distro)
  File "/usr/lib/python3/dist-packages/rosdistro/__init__.py", line 120, in get_distribution_file
    return create_distribution_file(dist_name, data)
  File "/usr/lib/python3/dist-packages/rosdistro/distribution_file.py", line 134, in create_distribution_file
    dist_file = DistributionFile(dist_name, d)
  File "/usr/lib/python3/dist-packages/rosdistro/distribution_file.py", line 63, in __init__
    self._add_package(pkg_name, repo)
  File "/usr/lib/python3/dist-packages/rosdistro/distribution_file.py", line 112, in _add_package
    assert pkg_name not in self.release_packages, "Duplicate package name '%s' exists in repository '%s' as well as in repository '%s'" % (pkg_name, repo.name, self.release_packages[pkg_name].repository_name)
AssertionError: Duplicate package name 'clpe' exists in repository 'clpe_sdk' as well as in repository 'clpe'
```

jonbonne, 2022-06-10:
same issue, ubuntu 20.04

`#27 14.78 
#27 14.78 ERROR: Rosdep experienced an error: Duplicate package name 'clpe' exists in repository 'clpe_sdk' as well as in repository 'clpe'
#27 14.78 Please go to the rosdep page [1] and file a bug report with the stack trace below.
#27 14.78 [1] : http://www.ros.org/wiki/rosdep
#27 14.78 
#27 14.78 rosdep version: 0.21.0
#27 14.78 
#27 14.78 Traceback (most recent call last):
#27 14.78   File "/usr/local/lib/python3.8/dist-packages/rosdep2/main.py", line 146, in rosdep_main
#27 14.78     exit_code = _rosdep_main(args)
#27 14.78   File "/usr/local/lib/python3.8/dist-packages/rosdep2/main.py", line 441, in _rosdep_main
#27 14.78     return _no_args_handler(command, parser, options, args)
#27 14.78   File "/usr/local/lib/python3.8/dist-packages/rosdep2/main.py", line 450, in _no_args_handler
#27 14.78     return command_handlers[command](options)
#27 14.78   File "/usr/local/lib/python3.8/dist-packages/rosdep2/main.py", line 659, in command_update
#27 14.78     update_sources_list(success_handler=update_success_handler,
#27 14.78   File "/usr/local/lib/python3.8/dist-packages/rosdep2/sources_list.py", line 516, in update_sources_list
#27 14.78     rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
#27 14.78   File "/usr/local/lib/python3.8/dist-packages/rosdep2/gbpdistro_support.py", line 149, in get_gbprepo_as_rosdep_data
#27 14.78     distro_file = get_release_file(gbpdistro)
#27 14.78   File "/usr/local/lib/python3.8/dist-packages/rosdep2/rosdistrohelper.py", line 76, in get_release_file
#27 14.78     dist_file = rosdistro.get_distribution_file(get_index(), distro)
#27 14.78   File "/usr/local/lib/python3.8/dist-packages/rosdistro/__init__.py", line 120, in get_distribution_file
#27 14.78     return create_distribution_file(dist_name, data)
#27 14.78   File "/usr/local/lib/python3.8/dist-packages/rosdistro/distribution_file.py", line 134, in create_distribution_file
#27 14.78     dist_file = DistributionFile(dist_name, d)
#27 14.78   File "/usr/local/lib/python3.8/dist-packages/rosdistro/distribution_file.py", line 63, in __init__
#27 14.78     self._add_package(pkg_name, repo)
#27 14.78   File "/usr/local/lib/python3.8/dist-packages/rosdistro/distribution_file.py", line 112, in _add_package
#27 14.78     assert pkg_name not in self.release_packages, "Duplicate package name '%s' exists in repository '%s' as well as in repository '%s'" % (pkg_name, repo.name, self.release_packages[pkg_name].repository_name)
#27 14.78 AssertionError: Duplicate package name 'clpe' exists in repository 'clpe_sdk' as well as in repository 'clpe'`

adriel-cobalt, 2022-06-10:
Same issue on Ubuntu 18.04

fyandun, 2022-06-10:
Same problem here. Ubuntu 18.04.6 in a Jetson Xavier dev kit.

rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Skip end-of-life distro "dashing"
Skip end-of-life distro "eloquent"
Add distro "foxy"

ERROR: Rosdep experienced an error: Duplicate package name 'clpe' exists in repository 'clpe_sdk' as well as in repository 'clpe'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.21.0

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 146, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 441, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 450, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 662, in command_update
    ros_distro=options.ros_distro)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 516, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python2.7/dist-packages/rosdep2/gbpdistro_support.py", line 149, in get_gbprepo_as_rosdep_data
    distro_file = get_release_file(gbpdistro)
  File "/usr/lib/python2.7/dist-packages/rosdep2/rosdistrohelper.py", line 76, in get_release_file
    dist_file = rosdistro.get_distribution_file(get_index(), distro)
  File "/usr/lib/python2.7/dist-packages/rosdistro/__init__.py", line 120, in get_distribution_file
    return create_distribution_file(dist_name, data)
  File "/usr/lib/python2.7/dist-packages/rosdistro/distribution_file.py", line 134, in create_distribution_file
    dist_file = DistributionFile(dist_name, d)
  File "/usr/lib/python2.7/dist-packages/rosdistro/distribution_file.py", line 63, in __init__
    self._add_package(pkg_name, repo)
  File "/usr/lib/python2.7/dist-packages/rosdistro/distribution_file.py", line 112, in _add_package
    assert pkg_name not in self.release_packages, "Duplicate package name '%s' exists in repository '%s' as well as in repository '%s'" % (pkg_name, [repo.name](http://repo.name/), self.release_packages[pkg_name].repository_name)
AssertionError: Duplicate package name 'clpe' exists in repository 'clpe_sdk' as well as in repository 'clpe'

mikeferguson, 2022-06-10:
It appears that the latest merge to rosdistro caused this - I've opened a PR to fix here: https://github.com/ros/rosdistro/pull/33536

zwilcox, 2022-06-11:
Thanks for the quick response.

verified issue is resolved:
```
 rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Skip end-of-life distro "dashing"
Skip end-of-life distro "eloquent"
Add distro "foxy"
Add distro "galactic"
Skip end-of-life distro "groovy"
Add distro "humble"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Skip end-of-life distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
Add distro "rolling"
updated cache in /home/some/path/.ros/rosdep/sources.cache
```

cottsay, 2022-06-11:
This should be resolved.

Follow-up issue to update the automation to prevent it in the future: ros/rosdistro#33537

closed 2022-06-11 by cottsay