ros-infrastructure/rosdep #889: Rosdep experienced an error: [Error 13]

TRo202, 2022-06-26:
Hi. I'm installing Moveit2 (ROS Foxy), and faced this error while following the installation guides on https://moveit.picknik.ai/foxy/doc/getting_started/getting_started.html

I'm struggling so much with installing Moveit2 for whole day, and I really hope I could get some help....

rosdep version: 0.22.1

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 146, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 446, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 542, in _package_args_handler
    lookup = _get_default_RosdepLookup(options)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 134, in _get_default_RosdepLookup
    sources_loader = SourcesListLoader.create_default(sources_cache_dir=options.sources_cache_dir,
  File "/usr/lib/python3/dist-packages/rosdep2/sources_list.py", line 601, in create_default
    sources = load_cached_sources_list(sources_cache_dir=sources_cache_dir, verbose=verbose)
  File "/usr/lib/python3/dist-packages/rosdep2/sources_list.py", line 563, in load_cached_sources_list
    return parse_sources_data(cache_data, origin=cache_index, model=model)
  File "/usr/lib/python3/dist-packages/rosdep2/sources_list.py", line 380, in parse_sources_data
    sources.append(model(type_, url, tags, origin=origin))
  File "/usr/lib/python3/dist-packages/rosdep2/sources_list.py", line 194, in create_model
    with open(pickle_filepath, 'rb') as f:
PermissionError: [Errno 13] Permission denied: '/home/tro/.ros/rosdep/sources.cache/7cfbeed211a3f9a6507c86cebaeedb3e3c4387f8.pickle'

christophebedard, 2022-06-28:
I think the above should solve your issue, so I'll close this.

If it still doesn't work, I'd encourage you to open a question on https://answers.ros.org/

closed 2022-06-28 by christophebedard

TRo202, 2022-06-29:
> I think the above should solve your issue, so I'll close this.
> 
> If it still doesn't work, I'd encourage you to open a question on https://answers.ros.org/

Thank you for your reply. It still doesn't solve the problem even I use `rosdep fix-permissions`. 

I think I should open a question on answers.ros

christophebedard, 2022-06-29:
> I think I should open a question on answers.ros

For future reference: https://answers.ros.org/question/403100/rosdep-permission-error-error-13/