ros-infrastructure/rosdep #890: problems running rosdep on raspberry pi

devangel77b, 2022-08-05:
it seems to be confused about the OS version. i tried running rosdep and also rosdep update -ay --os=debian or etc. it gave the following:

Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.22.1

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/rosdep2/main.py", line 146, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/home/pi/.local/lib/python3.9/site-packages/rosdep2/main.py", line 444, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/home/pi/.local/lib/python3.9/site-packages/rosdep2/main.py", line 453, in _no_args_handler
    return command_handlers[command](options)
  File "/home/pi/.local/lib/python3.9/site-packages/rosdep2/main.py", line 664, in command_update
    update_sources_list(success_handler=update_success_handler,
  File "/home/pi/.local/lib/python3.9/site-packages/rosdep2/sources_list.py", line 508, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/home/pi/.local/lib/python3.9/site-packages/rosdep2/gbpdistro_support.py", line 148, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/home/pi/.local/lib/python3.9/site-packages/rosdep2/__init__.py", line 91, in create_default_installer_context
    m.register_platforms(context)
  File "/home/pi/.local/lib/python3.9/site-packages/rosdep2/platforms/arch.py", line 53, in register_platforms
    register_manjaro(context)
  File "/home/pi/.local/lib/python3.9/site-packages/rosdep2/platforms/arch.py", line 58, in register_manjaro
    (os_name, os_version) = context.get_os_name_and_version()
  File "/home/pi/.local/lib/python3.9/site-packages/rosdep2/installers.py", line 115, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/home/pi/.local/lib/python3.9/site-packages/rospkg/os_detect.py", line 634, in get_name
    self.detect_os()
  File "/home/pi/.local/lib/python3.9/site-packages/rospkg/os_detect.py", line 604, in detect_os
    raise OsNotDetected("Could not detect OS, tried %s" % attempted)
rospkg.os_detect.OsNotDetected: Could not detect OS, tried ['zorin', 'windows', 'nixos', 'clearlinux', 'ubuntu', 'slackware', 'rocky', 'rhel', 'qnx', 'pop', 'osx', 'sailfishos', 'tizen', 'conda', 'oracle', 'opensuse', 'opensuse', 'opensuse', 'opensuse', 'opensuse', 'openembedded', 'neon', 'mx', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora', 'elementary', 'elementary', 'debian', 'cygwin', 'euleros', 'centos', 'manjaro', 'buildroot', 'arch', 'amazon', 'alpine', 'almalinux']

cottsay, 2022-08-05:
What version of `rospkg` are you using?

akela1101, 2022-08-27:
I had the same problem, and the answer to the question above is `1.4.0-1`.
```
File "/usr/lib/python3/dist-packages/rospkg/os_detect.py", line 604, in detect_os
    raise OsNotDetected("Could not detect OS, tried %s" % attempted)

$ dpkg -S /usr/lib/python3/dist-packages/rospkg/os_detect.py
python3-rospkg-modules: /usr/lib/python3/dist-packages/rospkg/os_detect.py

$ apt list | grep rospkg
python3-rospkg-modules/bullseye,now 1.4.0-1 all [installed,automatic]
```

#51 gives a hit to workaround this issue.
```
export ROS_OS_OVERRIDE=debian:bullseye
```
// Though the 51 is about `--os=debian:bullseye` option failure, which I can still confirm.

The problem is probably in os naming resolution (`raspbian` instead of `debian`)...?

mkatliar, 2022-11-28:
I have the same problem of Raspbian:

```
$ rosdep -v --os=debian:bullseye update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Skip end-of-life distro "dashing"
Skip end-of-life distro "eloquent"
Add distro "foxy"

ERROR: Rosdep experienced an error: Could not detect OS, tried ['zorin', 'windows', 'nixos', 'clearlinux', 'ubuntu', 'slackware', 'rhel', 'qnx', 'pop', 'osx', 'sailfishos', 'tizen', 'opensuse', 'opensuse', 'opensuse', 'opensuse', 'openembedded', 'neon', 'mx', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora', 'elementary', 'elementary', 'debian', 'cygwin', 'euleros', 'centos', 'manjaro', 'buildroot', 'arch', 'alpine']
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.20.0

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 144, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 428, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 437, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 646, in command_update
    update_sources_list(success_handler=update_success_handler,
  File "/usr/lib/python3/dist-packages/rosdep2/sources_list.py", line 516, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python3/dist-packages/rosdep2/gbpdistro_support.py", line 150, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/usr/lib/python3/dist-packages/rosdep2/__init__.py", line 91, in create_default_installer_context
    m.register_platforms(context)
  File "/usr/lib/python3/dist-packages/rosdep2/platforms/debian.py", line 65, in register_platforms
    register_elementary(context)
  File "/usr/lib/python3/dist-packages/rosdep2/platforms/debian.py", line 95, in register_elementary
    (os_name, os_version) = context.get_os_name_and_version()
  File "/usr/lib/python3/dist-packages/rosdep2/installers.py", line 115, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/usr/lib/python3/dist-packages/rospkg/os_detect.py", line 710, in get_name
    self.detect_os()
  File "/usr/lib/python3/dist-packages/rospkg/os_detect.py", line 680, in detect_os
    raise OsNotDetected("Could not detect OS, tried %s" % attempted)
rospkg.os_detect.OsNotDetected: Could not detect OS, tried ['zorin', 'windows', 'nixos', 'clearlinux', 'ubuntu', 'slackware', 'rhel', 'qnx', 'pop', 'osx', 'sailfishos', 'tizen', 'opensuse', 'opensuse', 'opensuse', 'opensuse', 'openembedded', 'neon', 'mx', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora', 'elementary', 'elementary', 'debian', 'cygwin', 'euleros', 'centos', 'manjaro', 'buildroot', 'arch', 'alpine']
```

The issue seems to be related to #11, as explicitly overriding OS version from the command line does not help.

The output of `lsb_release`:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```

mkatliar, 2022-11-28:
I have the same problem of Raspbian:

```
$ rosdep -v --os=debian:bullseye update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Skip end-of-life distro "dashing"
Skip end-of-life distro "eloquent"
Add distro "foxy"

ERROR: Rosdep experienced an error: Could not detect OS, tried ['zorin', 'windows', 'nixos', 'clearlinux', 'ubuntu', 'slackware', 'rhel', 'qnx', 'pop', 'osx', 'sailfishos', 'tizen', 'opensuse', 'opensuse', 'opensuse', 'opensuse', 'openembedded', 'neon', 'mx', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora', 'elementary', 'elementary', 'debian', 'cygwin', 'euleros', 'centos', 'manjaro', 'buildroot', 'arch', 'alpine']
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.20.0

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 144, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 428, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 437, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 646, in command_update
    update_sources_list(success_handler=update_success_handler,
  File "/usr/lib/python3/dist-packages/rosdep2/sources_list.py", line 516, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python3/dist-packages/rosdep2/gbpdistro_support.py", line 150, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/usr/lib/python3/dist-packages/rosdep2/__init__.py", line 91, in create_default_installer_context
    m.register_platforms(context)
  File "/usr/lib/python3/dist-packages/rosdep2/platforms/debian.py", line 65, in register_platforms
    register_elementary(context)
  File "/usr/lib/python3/dist-packages/rosdep2/platforms/debian.py", line 95, in register_elementary
    (os_name, os_version) = context.get_os_name_and_version()
  File "/usr/lib/python3/dist-packages/rosdep2/installers.py", line 115, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/usr/lib/python3/dist-packages/rospkg/os_detect.py", line 710, in get_name
    self.detect_os()
  File "/usr/lib/python3/dist-packages/rospkg/os_detect.py", line 680, in detect_os
    raise OsNotDetected("Could not detect OS, tried %s" % attempted)
rospkg.os_detect.OsNotDetected: Could not detect OS, tried ['zorin', 'windows', 'nixos', 'clearlinux', 'ubuntu', 'slackware', 'rhel', 'qnx', 'pop', 'osx', 'sailfishos', 'tizen', 'opensuse', 'opensuse', 'opensuse', 'opensuse', 'openembedded', 'neon', 'mx', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora', 'elementary', 'elementary', 'debian', 'cygwin', 'euleros', 'centos', 'manjaro', 'buildroot', 'arch', 'alpine']
```

The issue seems to be related to #51, as explicitly overriding OS version from the command line does not help.

The output of `lsb_release`:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```