ros-infrastructure/rosdep #891: Interest in development of support for authentication in rosdep

progtologist, 2022-09-05:
Apt has support for authentication for private APT repository servers.
I was wondering if I were to build such (or similar) functionality for rosdep, would this be something that the maintainers accept in a pull request or is this something that is beyond the scope of OSRF/rosdep?

AFAIK only authentication for proxies is supported which is different from what APT offers.

nuclearsandwich, 2022-09-05:
Support for apt repositories which require authentication/authorization is not something that the ROS Infrastructure team anticipates being on our roadmap which means that this support would need to be driven by community interest.

rosdep currently interacts with system apt repositories by invoking existing command line tools. 
My first question is to ask whether rosdep _needs_ any special support or if configuring authorization for a specific repository could be done within apt and not require any special handling in rosdep. If explicit authentication handling is required, my initial expectations for a pull request from the community to be reviewable would include:

* Links to documentation describing the authorization/authentication interface being implemented.
* A plan for automated testing of the authenticated code flow since it will need to be maintained by the rosdep team.

progtologist, 2022-09-06:
Ah, my bad, reading through my message now I understand how it could be confusing. I am not referring on rosdep supporting authenticated/authorized APT repos - this it already supports given as you said it doesn't interact directly with the repos itself but through the distribution's tools.

My question is more in like with supporting rosdep source lists that are behind a server that requires authentication.
E.g. adding an extra file under `/etc/ros/rosdep/sources.list.d/` that would require an entry such as:
```yaml
yaml https://user:password@server.com/ros/rosdep.yaml
```
Currently, rosdep when parsing this makes the assumption that everything after `:` is the port number.

cottsay, 2022-09-07:
This is a pretty reasonable request. I hashed out a preliminary PR in #892. Please give that a shot and provide feedback if it works for you.

progtologist, 2022-09-14:
@cottsay have a look at #895 and let me know what you think!