ros-infrastructure/rosdep #894: ERROR: Rosdep experienced an error: [Errno 2] No such file or directory: '/home/rosdev/ws_moveit2/install/rviz_marker_tools/share/rviz_marker_tools/package.xml'

enriLoniterp, 2022-09-12:
Hi, rosdep initied this projetct and got this error telling me to report on bug features.

```
ERROR: Rosdep experienced an error: [Errno 2] No such file or directory: '/home/rosdev/ws_moveit2/install/rviz_marker_tools/share/rviz_marker_tools/package.xml'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.22.1

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 146, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 446, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 555, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 764, in command_install
    uninstalled, errors = installer.get_uninstalled(packages, implicit=options.recursive, verbose=options.verbose)
  File "/usr/lib/python3/dist-packages/rosdep2/installers.py", line 445, in get_uninstalled
    resolutions, errors = self.lookup.resolve_all(resources, installer_context, implicit=implicit)
  File "/usr/lib/python3/dist-packages/rosdep2/lookup.py", line 406, in resolve_all
    rosdep_keys = self.get_rosdeps(resource_name, implicit=implicit)
  File "/usr/lib/python3/dist-packages/rosdep2/lookup.py", line 324, in get_rosdeps
    return self.loader.get_rosdeps(resource_name, implicit=implicit)
  File "/usr/lib/python3/dist-packages/rosdep2/rospkg_loader.py", line 144, in get_rosdeps
    if resource_name in self.get_catkin_paths():
  File "/usr/lib/python3/dist-packages/rosdep2/rospkg_loader.py", line 133, in get_catkin_paths
    self._catkin_packages_cache = dict(find_catkin_paths(self._rospack))
  File "/usr/lib/python3/dist-packages/rosdep2/rospkg_loader.py", line 132, in find_catkin_paths
    filter(lambda x: src.get_manifest(x).is_catkin, src.list())]
  File "/usr/lib/python3/dist-packages/rospkg/rospack.py", line 196, in list
    self._update_location_cache()
  File "/usr/lib/python3/dist-packages/rospkg/rospack.py", line 188, in _update_location_cache
    list_by_path(self._manifest_name, path, cache)
  File "/usr/lib/python3/dist-packages/rospkg/rospack.py", line 71, in list_by_path
    root = ElementTree(None, os.path.join(d, PACKAGE_FILE))
  File "/usr/lib/python3.10/xml/etree/ElementTree.py", line 540, in __init__
    self.parse(file)
  File "/usr/lib/python3.10/xml/etree/ElementTree.py", line 569, in parse
    source = open(source, "rb")
FileNotFoundError: [Errno 2] No such file or directory: '/home/rosdev/ws_moveit2/install/rviz_marker_tools/share/rviz_marker_tools/package.xml'
```

nuclearsandwich, 2022-09-12:
Thanks for the report. Could you also share the rosdep command that you were running when you encountered the error.

enriLoniterp, 2022-09-12:
yes of course,

I was running 

sudo apt update &&` rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y`

nuclearsandwich, 2022-09-12:
I had started to write a bunch more directed questions but each of them has an assumption about what your setup is.

In addition to the command could you provide some additional context. Are you building moveit2? Using a pre-build moveit2 workspace to build more on top or just trying to use it? Are you following any specific guides or tutorials.

Was the current path the workspace root (I'm guessing `/home/rosdev/ws_moveit2` from the stacktrace) or was it performed from the `src` subdirectory of the workspace root?

Is there a file at `/home/rosdev/ws_moveit2/install/rviz_marker_tools/share/rviz_marker_tools/package.xml`?

In short, I'm rather confused by this error since it appears to be a disconnect between the detection of a catkin package and the presence of a package.xml where one is expected so figuring out the environment that replicates this issue will be what helps us diagnose it.

enriLoniterp, 2022-09-13:
hi, i'm just following the tutorial to build moveit project as described in the official page.
But it second time i used rosdep command in the **root** of my workspace. 

Thank you for the suggestion, i've seen that package.xml is a link broken:
Link (broken) (inode/symlink).

Could it be something usefull?

Hope i'm helping!

ralwing, 2023-04-19:
I just faced the same problem:
``` 
 apt-get update -y     && rosdep update &&   rosdep install --from-paths /ws/src --ignore-packages-from-source -y --skip-keys "as_ros_test_utils"
Hit:1 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2204/x86_64  InRelease
Hit:2 http://archive.ubuntu.com/ubuntu jammy InRelease              
Hit:3 http://packages.ros.org/ros2/ubuntu jammy InRelease           
Hit:4 http://security.ubuntu.com/ubuntu jammy-security InRelease
Hit:5 http://archive.ubuntu.com/ubuntu jammy-updates InRelease
Hit:6 http://archive.ubuntu.com/ubuntu jammy-backports InRelease
Reading package lists... Done
reading in sources list data from /etc/ros/rosdep/sources.list.d
Warning: running 'rosdep update' as root is not recommended.
  You should run 'sudo rosdep fix-permissions' and invoke 'rosdep update' again without sudo.
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Skip end-of-life distro "dashing"
Skip end-of-life distro "eloquent"
Add distro "foxy"
Skip end-of-life distro "galactic"
Skip end-of-life distro "groovy"
Add distro "humble"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Skip end-of-life distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
Add distro "rolling"
updated cache in /root/.ros/rosdep/sources.cache

ERROR: Rosdep experienced an error: [Errno 2] No such file or directory: '/ws/src/as_drive_nodes/install/as_ros_test_utils/share/as_ros_test_utils/package.xml'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.22.2

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 146, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 446, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 555, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 765, in command_install
    uninstalled, errors = installer.get_uninstalled(packages, implicit=options.recursive, verbose=options.verbose)
  File "/usr/lib/python3/dist-packages/rosdep2/installers.py", line 445, in get_uninstalled
    resolutions, errors = self.lookup.resolve_all(resources, installer_context, implicit=implicit)
  File "/usr/lib/python3/dist-packages/rosdep2/lookup.py", line 406, in resolve_all
    rosdep_keys = self.get_rosdeps(resource_name, implicit=implicit)
  File "/usr/lib/python3/dist-packages/rosdep2/lookup.py", line 324, in get_rosdeps
    return self.loader.get_rosdeps(resource_name, implicit=implicit)
  File "/usr/lib/python3/dist-packages/rosdep2/rospkg_loader.py", line 144, in get_rosdeps
    if resource_name in self.get_catkin_paths():
  File "/usr/lib/python3/dist-packages/rosdep2/rospkg_loader.py", line 133, in get_catkin_paths
    self._catkin_packages_cache = dict(find_catkin_paths(self._rospack))
  File "/usr/lib/python3/dist-packages/rosdep2/rospkg_loader.py", line 132, in find_catkin_paths
    filter(lambda x: src.get_manifest(x).is_catkin, src.list())]
  File "/usr/lib/python3/dist-packages/rospkg/rospack.py", line 196, in list
    self._update_location_cache()
  File "/usr/lib/python3/dist-packages/rospkg/rospack.py", line 188, in _update_location_cache
    list_by_path(self._manifest_name, path, cache)
  File "/usr/lib/python3/dist-packages/rospkg/rospack.py", line 71, in list_by_path
    root = ElementTree(None, os.path.join(d, PACKAGE_FILE))
  File "/usr/lib/python3.10/xml/etree/ElementTree.py", line 540, in __init__
    self.parse(file)
  File "/usr/lib/python3.10/xml/etree/ElementTree.py", line 569, in parse
    source = open(source, "rb")
FileNotFoundError: [Errno 2] No such file or directory: '/ws/src/as_drive_nodes/install/as_ros_test_utils/share/as_ros_test_utils/package.xml'
```
This is kind of strange because it happens only on my computer, running this on our CI works fine.

ralwing, 2023-04-19:
I've increased verbosity by printing function arguments:
``` 
File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 146, in rosdep_main
    exit_code = _rosdep_main(args)
    args = ['install', '--from-paths', '/ws/src', '--ignore-packages-from-source', '-y']
    e = FileNotFoundError(2, 'No such file or directory')
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 448, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
    args = ['/ws/src']
    command = 'install'
    default_sources_cache = '/root/.ros/rosdep/sources.cache'
    options = <Values at 0x7f8a5ed37280: {'os_override': None, 'sources_cache_dir': '/root/.ros/rosdep/sources.cache', 'verbose': False, 'print_version': False, 'print_all_versions': False, 'reinstall': False, 'default_yes': True, 'simulate': False, 'robust': False, 'quiet': False, 'rosdep_all': False, 'recursive': True, 'ignore_src': True, 'skip_keys': [], 'filter_for_installers': [], 'from_paths': True, 'ros_distro': None, 'as_root': {}, 'include_eol_distros': False, 'dependency_types': []}>
    parser = <optparse.OptionParser object at 0x7f8a5fe731f0>
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 557, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
    args = ['/ws/src']
    command = 'install'
    lookup = <rosdep2.lookup.RosdepLookup object at 0x7f8a5eeaa350>
    not_found = []
    options = <Values at 0x7f8a5ed37280: {'os_override': None, 'sources_cache_dir': '/root/.ros/rosdep/sources.cache', 'verbose': False, 'print_version': False, 'print_all_versions': False, 'reinstall': False, 'default_yes': True, 'simulate': False, 'robust': False, 'quiet': False, 'rosdep_all': False, 'recursive': True, 'ignore_src': True, 'skip_keys': [], 'filter_for_installers': [], 'from_paths': True, 'ros_distro': None, 'as_root': {}, 'include_eol_distros': False, 'dependency_types': []}>
    packages = ['as_point_cloud_merger', 'swri_cli_tools', 'as_trajectory_viz_throttle', 'robot_localization', 'fixposition_gnss_tf', 'as_ros_license', 'ez10_driver', 'as_stilo_can_to_odometry', 'as_point_cloud_localizer', 'as_imu_diagnostic', 'swri_serial_util', 'as_lanelet_extensions', 'as_fmt_helpers', 'as_2d_to_3d_transformers', 'tier4_planning_msgs', 'as_geometry', 'tier4_external_api_msgs', 'as_static_camera_info_publisher', 'tier4_system_msgs', 'as_lanelet_map_provider', 'tier4_vehicle_msgs', 'as_stop_validator', 'as_bb1_regulator_model', 'as_rgb_to_rccb', 'as_trajectory_smoother', 'as_motion_model', 'as_dynamic_objects_costmap', 'as_ros_autoware_tf_utils', 'as_bb1_driver', 'as_imu_orientation', 'as_amd_vehicle_state_publisher', 'novatel_gps_driver', 'as_launch_extensions', 'as_autonomy_controller', 'as_time', 'as_gnss_to_base_link', 'as_pcr_benchmark', 'as_dynamic_objects_costmap_nav2_plugin', 'as_elevation_pose', 'as_test_utils', 'as_pose_predictor', 'as_lidar_semantic_segmentation_viz', 'as_local_planner', 'bounding_box_frame_transform_node', 'as_colorized_segmentation_to_labeled', 'swri_roscpp', 'as_ros_autoware_msg_utils', 'as_odometry_to_tf2', 'as_static_orientation', 'as_ros_test_utils', 'as_ros_utils', 'tier4_auto_msgs_converter', 'driveworks_msgs', 'as_odom_to_pose', 'as_velodyne', 'swri_geometry_util', 'novatel_gps_msgs', 'tier4_localization_msgs', 'tier4_v2x_msgs', 'tier4_rtc_msgs', 'ez10_driver_mock', 'swri_prefix_tools', 'as_covariance_insertion', 'swri_transform_util', 'as_point_cloud_merge_gpu', 'as_freespace_3d_wall', 'as_can_odometry_converter', 'as_object_collision_estimator_nodes', 'as_driveworks_plugins', 'as_kalman', 'as_component_tests', 'as_path_validator', 'as_lla_to_ecef', 'swri_opencv_util', 'as_costmap_common', 'swri_route_util', 'socket_can', 'swri_system_util', 'swri_console_util', 'as_object_collision_estimator', 'as_drive_msgs', 'as_compass', 'swri_image_util', 'as_monitor_analyzers', 'tier4_hmi_msgs', 'as_traits', 'as_drive_cmake', 'as_rviz_plugins', 'tier4_control_msgs', 'as_map_loader', 'as_topic_monitor', 'as_eigen', 'tier4_simulation_msgs', 'as_metrics', 'tier4_debug_msgs', 'as_odometry_to_vehicle_kinematic_state', 'as_bb1_regulator', 'as_novatel_gnss_covariance', 'as_ros_common', 'as_pure_pursuit_extension', 'as_color_maps', 'as_i_localization_filter', 'as_net_drivers', 'as_tf_namespace_mapper', 'swri_dbw_interface', 'as_collision_path_throttler', 'as_exception', 'as_math', 'as_ros_tf_utils', 'as_scripts', 'as_heartbeat_mock', 'as_fundament_blocks', 'as_pose_provider', 'ndt_omp', 'odometry_generator', 'as_earth_localization', 'tier4_api_msgs', 'ez10_to_twist', 'as_messages', 'as_ros_diagnostic_utils', 'as_conversions', 'as_linear_accelerations', 'as_vehicle_ctrl_cmd_to_ackermann_ctrl_cmd', 'as_ez10_vehicle_status_converter', 'as_gps_to_map', 'tier4_perception_msgs', 'as_vehicle_info', 'swri_math_util', 'as_ros_msg_utils', 'as_ublox_gnss_covariance', 'as_navsat_diagnostic', 'ctrl_cmd_to_ackermann_drive', 'as_metrics_agent', 'as_map_manager', 'as_dead_reckoning_odometry', 'as_ros_parameters', 'as_ublox_msgs_to_diagnostic', 'as_global_planner', 'as_point_cloud_unifier', 'as_logger', 'as_carla_add_sensor_covariance', 'as_rosbag_reader']
    parser = <optparse.OptionParser object at 0x7f8a5fe731f0>
    path = '/ws/src'
    pkgs = dict_keys(['actionlib_msgs', 'rosbag2', 'rosbag2_storage_default_plugins', 'ament_cmake_export_link_flags', 'ros2cli', 'sensor_msgs', 'ament_index_cpp', 'ament_cmake_export_dependencies', 'launch_testing_ament_cmake', 'ament_lint_common', ' [... cut ...]
    self = <xml.etree.ElementTree.ElementTree object at 0x7f8a5ee91bd0>
  File "/usr/lib/python3.10/xml/etree/ElementTree.py", line 569, in parse
    source = open(source, "rb")
    close_source = False
    parser = None
    self = <xml.etree.ElementTree.ElementTree object at 0x7f8a5ee91bd0>
    source = '/ws/src/as_drive_nodes/install/as_ros_test_utils/share/as_ros_test_utils/package.xml'
FileNotFoundError: [Errno 2] No such file or directory: '/ws/src/as_drive_nodes/install/as_ros_test_utils/share/as_ros_test_utils/package.xml'

```

ralwing, 2023-04-19:
Seems like the problem lies in existance of `install` folder under the `src` folder.  The `as_ros_test_utils` is then seen twice.