ros-infrastructure/rosdep #898: 'utf-8' codec can't decode byte 0xe0 in position 39: invalid continuation byte

alain-lnpcrd, 2022-09-22:
ROS2 on raspberrypi os Bullseye

`pi@raspberrypi:~/ros2_ws $ rosdep install -i --from-path src --rosdistro humble -y

ERROR: Rosdep experienced an internal error.
Please go to the rosdep page [1] and file a bug report with the message below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.22.1

Bad installer [apt]: 'utf-8' codec can't decode byte 0xe0 in position 39: invalid continuation byte

pi@raspberrypi:~/ros2_ws $ `

ijnek, 2022-09-23:
I'm guessing this is related to not having run the instructions in [Set locale](https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html#set-locale). Does the issue persist after running those commands?

christophebedard, 2022-09-23:
> Could you tell us whether the issue persists after running those commands?

If that fixes it, maybe we could modify `rosdep` to catch that error and print a message/hint about setting the locale?

alain-lnpcrd, 2022-09-24:
I gave up raspberrypi os because with the procedure found the installation is very incomplete.
In fact the problem with ubuntu is that it lacks python3-libcamera and I want to rewrite the python program of an "AWR-Adeept" robot car toy with ROS2 nodes. Currently I run it under raspberrypi OS aarch64 with libcamera for video. If I add "archive.raspberrypi.org bullseye" in repositories as found as solution, python3-libcamera works with python < 3.10 while python3 jammy is 3.10. In addition, a bug appears with raspberrypi-kernel which is installed and the session display ends up bugging.