ros-infrastructure/rosdep #904: rosdep install should have an --update flag to refresh package cache before performing install

codygarver, 2022-10-27:
Our use case is in CI we use containers whose apt package cache is often stale, causing `rosdep install` to eventually fail unless `apt-get update` has run recently.

I propose that `rosdep install` should have an `--update` flag that will refresh the package cache if the backend package manager supports that. The equivalent of running `apt-get update && rosdep install foo`.

peci1, 2023-01-02:
I don't think this is a job for rosdep.

You should ensure re-validation of the APT cache yourself. In docker, you'd write:

    RUN apt update && rosdep install foo

And, ideally:

    RUN apt update && rosdep update && rosdep install foo

codygarver, 2023-01-03:
Since rosdep is acting as a wrapper for `apt install foo`, I was thinking it made sense to wrap the requirement for that underlying command to succeed. However, I'm pretty new to using rosdep at all so I'll take your word for it if that doesn't fit the intended use of rosdep.

peci1, 2023-01-03:
An example can be found e.g. here: https://github.com/osrf/subt/blob/master/docker/cloudsim_sim/Dockerfile#L105-L115 .

SubaruArai, 2024-04-10:
rosdep not only manages apt, but also other package managers, such as rpm, brew, etc.
While I agree adding an option to update the cache (e.g. `--update-cache`) would be nice, we'll need to add functions for all the package managers (which has OSs that I don't have access to).

tldr: it's quite complicated to implement properly, let alone testing it.