ros-infrastructure/rosdep #905: rosdep keys recursive?

tonynajjar, 2022-11-02:


tfoote, 2022-11-02:
rosdep does not have access to the recursive dependencies. That is only known by the package manager to which rosdep delegates the installation after resolving the keys and then basically reinvoking rosdep. But that's a lot of scope creep for the tool.

tonynajjar, 2022-11-02:
Thank you for your answer. I actually meant just the dependencies mentioned in the package.xml files, rosdep doesn't have access to that? I found a workaround for it:

```
root@26b51d9a5428:/code# rosdep keys $(colcon list --packages-up-to navigation2 -n)
libboost-program-options-dev
rcl_action
bondcpp
composition_interfaces
launch
ament_lint_common
nav_msgs
ament_lint_auto
diagnostic_updater
rclcpp_lifecycle
dwb_critics
libceres-dev
nav2_planner
nav2_util
performance_test_fixture
ament_cmake_python
rosidl_default_generators
behaviortree_cpp_v3
ament_cmake_gen_version_h
rcpputils
rmw_implementation
image_transport
ament_index_cpp
launch_testing
libqt5-opengl
rosidl_generator_py
rclcpp
rcl_interfaces
dwb_msgs
rcl_logging_interface
launch_ros
libqt5-widgets
nav2_smac_planner
nav_2d_utils
osrf_pycommon
ament_pep257
nlohmann-json-dev
python3-pytest
class_loader
pybind11_vendor
rviz_common
unique_identifier_msgs
map_msgs
tf2_geometry_msgs
nav2_velocity_smoother
std_srvs
eigen
nav2_msgs
nav2_theta_star_planner
ament_cmake_gtest
rclcpp_action
ament_flake8
ament_copyright
nav_2d_msgs
geometry_msgs
nav2_smoother
nav2_core
statistics_msgs
test_msgs
rcl_lifecycle
rclcpp_components
rviz_default_plugins
nav2_waypoint_follower
message_filters
dwb_core
visualization_msgs
python3
nav2_behavior_tree
ament_cmake_gmock
rosgraph_msgs
ament_cmake_ros
launch_testing_ament_cmake
laser_geometry
nav2_voxel_grid
nav2_behaviors
nav2_common
rclpy
builtin_interfaces
lifecycle_msgs
nav2_regulated_pure_pursuit_controller
pluginlib
rpyutils
action_msgs
python3-yaml
nav2_bt_navigator
tf2_ros
nav2_lifecycle_manager
rosidl_runtime_c
nav2_navfn_planner
python_cmake_module
graphicsmagick
nav2_rviz_plugins
rcl_yaml_param_parser
sensor_msgs
dwb_plugins
std_msgs
rviz_rendering
libboost-program-options
ament_cmake
nav2_dwb_controller
mimick_vendor
ament_cmake_pytest
rcl
tf2
rosidl_runtime_cpp
libqt5-core
ament_index_python
nav2_amcl
ament_cmake_core
bond
rmw_implementation_cmake
rosidl_typesupport_cpp
tf2_sensor_msgs
ament_cmake_google_benchmark
resource_retriever
nav2_costmap_2d
rmw
libqt5-gui
nav2_rotation_shim_controller
nav2_constrained_smoother
rosidl_default_runtime
ompl
qtbase5-dev
rosidl_typesupport_c
costmap_queue
nav2_map_server
cv_bridge
eigen3_cmake_module
rviz_ogre_vendor
tracetools
launch_testing_ros
rcutils
libstatistics_collector
nav2_controller
angles
yaml_cpp_vendor
nav2_simple_commander
```

tonynajjar, 2022-11-03:
I'll go ahead and close this since I found a workaround, thanks for your quick reply. You can reopen it if you think the workaround is something rosdep could support by itself

closed 2022-11-03 by tonynajjar