ros-infrastructure/rosdep #906: rosdep init time out

Neptune-Ruijie, 2022-11-09:
root@nano:~/ros2_dashing$ sudo rosdep init

ERROR: Rosdep experienced an error: ('The read operation timed out',)
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.22.1

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 146, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 444, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 453, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 597, in command_init
    data = download_default_sources_list()
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 326, in download_default_sources_list
    f = urlopen_gzip(url, timeout=DOWNLOAD_TIMEOUT)
  File "/usr/lib/python2.7/dist-packages/rosdep2/url_utils.py", line 50, in urlopen_gzip
    response = urlopen(url_request, **kwargs)
  File "/usr/lib/python2.7/urllib2.py", line 154, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python2.7/urllib2.py", line 429, in open
    response = self._open(req, data)
  File "/usr/lib/python2.7/urllib2.py", line 447, in _open
    '_open', req)
  File "/usr/lib/python2.7/urllib2.py", line 407, in _call_chain
    result = func(*args)
  File "/usr/lib/python2.7/urllib2.py", line 1248, in https_open
    context=self._context)
  File "/usr/lib/python2.7/urllib2.py", line 1208, in do_open
    r = h.getresponse(buffering=True)
  File "/usr/lib/python2.7/httplib.py", line 1178, in getresponse
    response.begin()
  File "/usr/lib/python2.7/httplib.py", line 452, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python2.7/httplib.py", line 408, in _read_status
    line = self.fp.readline(_MAXLINE + 1)
  File "/usr/lib/python2.7/socket.py", line 480, in readline
    data = self._sock.recv(self._rbufsize)
  File "/usr/lib/python2.7/ssl.py", line 754, in recv
    return self.read(buflen)
  File "/usr/lib/python2.7/ssl.py", line 641, in read
    v = self._sslobj.read(len)
SSLError: ('The read operation timed out',)

nuclearsandwich, 2022-11-09:
Thanks for the report @Neptune-Ruijie. The message above looks like a network related issue rather than a problem with rosdep itself. Were you later able to run rosdep init successfully? If so can you please close this issue?

Neptune-Ruijie, 2022-11-30:
Thanks, it is a network issue. It will run successfully if you add a proxy to the netwoek.