ros-infrastructure/rosdep #917: Multiple packages found with the same name

mtgizaw15, 2023-03-30:
rosdep version: 0.22.2

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 146, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 446, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 501, in _package_args_handler
    pkgs = find_catkin_packages_in(path, options.verbose)
  File "/usr/lib/python3/dist-packages/rosdep2/catkin_packages.py", line 35, in find_catkin_packages_in
    packages = find_packages(path)
  File "/usr/lib/python3/dist-packages/catkin_pkg/packages.py", line 103, in find_packages
    raise RuntimeError('\n'.join(duplicates))
RuntimeError: Multiple packages found with the same name "depthai_examples":
- luxonis/depthai-ros-examples/depthai_examples
- luxonis/depthai-ros/depthai_examples

christophebedard, 2023-03-30:
This is not related to `rosdep` itself. You have packages with the same name, as shown in the error output:

* luxonis/depthai-ros-examples/depthai_examples
* luxonis/depthai-ros/depthai_examples

Remove one of the packages and try again.

closed 2023-03-30 by christophebedard

mtgizaw15, 2023-03-30:
Can you help me which command should I use to remove one of the packages because the "sudo apt-get remove luxonis/depthai-ros-examples/depthai_examples" not removing the package.

christophebedard, 2023-03-30:
I think these packages are in your workspace, so:

```
$ cd ws/
$ rm -rf luxonis/depthai-ros-examples/depthai_examples/
```

mtgizaw15, 2023-03-30:
Yes it was there thank you so much.