ros-infrastructure/rosdep #934: web site is down?

OverlordYashSrivastav1, 2023-10-07:
c:\Robot>rosdep init
ERROR: cannot download default sources list from:
https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
Website may be down.

sloretz, 2024-02-19:
I seem the same issue on `ubuntu:focal` in a github action. I can `curl` the URL, but `python3-rosdep` fails to download it

```
STEP 4/7: RUN curl -S https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100   588  100   588    0     0   3340      0 --:--:-- --:--:-- --:--:--  3520
# os-specific listings first
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml osx
# generic
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
gbpdistro https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml fuerte
# newer distributions (Groovy, Hydro, ...) must not be listed anymore, they are being fetched from the rosdistro index.yaml instead
STEP 5/7: RUN bash -c 'for i in {1..5}; do rosdep init && break || sleep 15; done'
ERROR: cannot download default sources list from:
https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
Website may be down.
```

sloretz, 2024-02-19:
I think it's coming from the `URLError` case here, since that case doesn't print the error message :(

https://github.com/ros-infrastructure/rosdep/blob/5ca91b8a645a9f26776db4538f254521f5a484d2/src/rosdep2/main.py#L604-L606

sloretz, 2024-02-19:
Forgive me for the screenshot of text, but I don't have copy/paste working in my VM. The root cause appears to be an SSL issue

[image]

sloretz, 2024-02-19:
Resolved on my machine by running `sudo update-ca-certificates --fresh` before `rosdep init` on ubuntu:focal armhf.