ros-infrastructure/rosdep #935: rosdep update failed in python virtual environment on macOS Sonoma 14.1.2

pyfb78, 2023-12-10:
```reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Skip end-of-life distro "dashing"
Skip end-of-life distro "eloquent"
Skip end-of-life distro "foxy"
Skip end-of-life distro "galactic"
Skip end-of-life distro "groovy"
Add distro "humble"

ERROR: Rosdep experienced an error: unrecognized version: 14
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.22.2

Traceback (most recent call last):
  File "/Users/pavan/.local/share/virtualenvs/ROS-aFsUwaBr/lib/python3.8/site-packages/rosdep2/main.py", line 146, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/Users/pavan/.local/share/virtualenvs/ROS-aFsUwaBr/lib/python3.8/site-packages/rosdep2/main.py", line 444, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/Users/pavan/.local/share/virtualenvs/ROS-aFsUwaBr/lib/python3.8/site-packages/rosdep2/main.py", line 453, in _no_args_handler
    return command_handlers[command](options)
  File "/Users/pavan/.local/share/virtualenvs/ROS-aFsUwaBr/lib/python3.8/site-packages/rosdep2/main.py", line 665, in command_update
    update_sources_list(success_handler=update_success_handler,
  File "/Users/pavan/.local/share/virtualenvs/ROS-aFsUwaBr/lib/python3.8/site-packages/rosdep2/sources_list.py", line 510, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/Users/pavan/.local/share/virtualenvs/ROS-aFsUwaBr/lib/python3.8/site-packages/rosdep2/gbpdistro_support.py", line 148, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/Users/pavan/.local/share/virtualenvs/ROS-aFsUwaBr/lib/python3.8/site-packages/rosdep2/__init__.py", line 91, in create_default_installer_context
    m.register_platforms(context)
  File "/Users/pavan/.local/share/virtualenvs/ROS-aFsUwaBr/lib/python3.8/site-packages/rosdep2/platforms/arch.py", line 53, in register_platforms
    register_manjaro(context)
  File "/Users/pavan/.local/share/virtualenvs/ROS-aFsUwaBr/lib/python3.8/site-packages/rosdep2/platforms/arch.py", line 58, in register_manjaro
    (os_name, os_version) = context.get_os_name_and_version()
  File "/Users/pavan/.local/share/virtualenvs/ROS-aFsUwaBr/lib/python3.8/site-packages/rosdep2/installers.py", line 115, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/Users/pavan/.local/share/virtualenvs/ROS-aFsUwaBr/lib/python3.8/site-packages/rospkg/os_detect.py", line 636, in get_name
    self.detect_os()
  File "/Users/pavan/.local/share/virtualenvs/ROS-aFsUwaBr/lib/python3.8/site-packages/rospkg/os_detect.py", line 598, in detect_os
    self._os_codename = os_detector.get_codename()
  File "/Users/pavan/.local/share/virtualenvs/ROS-aFsUwaBr/lib/python3.8/site-packages/rospkg/os_detect.py", line 353, in get_codename
    return _osx_codename(*ver[0:2])
  File "/Users/pavan/.local/share/virtualenvs/ROS-aFsUwaBr/lib/python3.8/site-packages/rospkg/os_detect.py", line 331, in _osx_codename
    raise OsNotDetected("unrecognized version: %s" % key)
rospkg.os_detect.OsNotDetected: unrecognized version: 14
```

kylegibson15, 2023-12-30:
Im experiencing the same error on macOS Sonoma 14.2.1

pyfb78, 2024-01-01:
Thank you for finding a solution to that. I was wondering if you figured out a way to deal with the issue that occres when you try to resolve all of the dependencies? When I run the command 
```
rosdep install --from-paths ./src --ignore-packages-from-source --rosdistro noetic -y
```
I get the following output as error
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
roslib: No definition of [libboost-filesystem-dev] for OS [osx]
message_filters: No definition of [libboost-thread-dev] for OS [osx]
catkin: No definition of [python3-catkin-pkg] for OS [osx]
rviz: No definition of [libogre] for OS [osx]
rosconsole: No definition of [libboost-regex-dev] for OS [osx]
rospy: No definition of [python3-numpy] for OS [osx]
rospack: No definition of [python3-coverage] for OS [osx]
diagnostic_common_diagnostics: No definition of [lm-sensors] for OS [osx]
actionlib: No definition of [libboost-thread-dev] for OS [osx]
rostest: No definition of [libboost-thread-dev] for OS [osx]
turtlesim: No definition of [libboost-thread-dev] for OS [osx]
roslaunch: No definition of [python3-paramiko] for OS [osx]
pluginlib: No definition of [libboost-filesystem-dev] for OS [osx]
tf_conversions: No definition of [python3-pykdl] for OS [osx]
cv_bridge: No definition of [python3-numpy] for OS [osx]
roscpp_tutorials: No definition of [libboost-thread-dev] for OS [osx]
class_loader: No definition of [libboost-system-dev] for OS [osx]
rqt_plot: No definition of [python3-numpy] for OS [osx]
python_qt_binding: No definition of [python3-qt5-bindings] for OS [osx]
gl_dependency: No definition of [python3-qt5-bindings-gl] for OS [osx]
rosgraph: No definition of [python3-netifaces] for OS [osx]
laser_geometry: No definition of [python3-numpy] for OS [osx]
rosmaster: No definition of [python3-defusedxml] for OS [osx]
tf2_geometry_msgs: No definition of [python3-pykdl] for OS [osx]
bondcpp: No definition of [libboost-thread-dev] for OS [osx]
rosbag_storage: No definition of [libgpgme-dev] for OS [osx]
roswtf: No definition of [python3-paramiko] for OS [osx]
qt_gui: No definition of [python3-qt5-bindings] for OS [osx]
rosbag: No definition of [libboost-thread-dev] for OS [osx]
xmlrpcpp: No definition of [libboost-thread-dev] for OS [osx]
dynamic_reconfigure: No definition of [libboost-dev] for OS [osx]
filters: No definition of [libboost-dev] for OS [osx]
rqt_pose_view: No definition of [python3-opengl] for OS [osx]
cpp_common: No definition of [libboost-thread-dev] for OS [osx]
rqt_bag_plugins: No definition of [python3-cairo] for OS [osx]
roscpp: No definition of [libboost-system-dev] for OS [osx]
genpy: No definition of [python3-numpy] for OS [osx]
rostime: No definition of [libboost-thread-dev] for OS [osx]
webkit_dependency: No definition of [python3-qt5-bindings-webkit] for OS [osx]
resource_retriever: No definition of [libcurl] for OS [osx]
```