ros-infrastructure/rosdep #940: Apt install one-shot

tonynajjar, 2024-02-09:
This is more of a suggestion than an issue. I noticed that rosdep runs each apt install individually for each package which is probably more time consuming than running 'apt install package1 package2....'. Can we install them all in one-shot? This small change could save a lot of CI time for a lot of people.

What do you think?

SubaruArai, 2024-04-15:
@tonynajjar Made a PR, care to test it out?

tonynajjar, 2024-04-15:
hi @SubaruArai, thanks for tackling this! I tested it out and for my use case it accelerated the rosdep installation by 30 seconds which is a big improvement!

I'm just curious about one thing I see in the logs:

```
#28 27.51 executing command [sudo -H apt-get install -y python3-imageio python3-mock python3-pytest-cov ros-humble-foxglove-bridge ros-humble-foxglove-msgs ros-humble-nav-2d-msgs ros-humble-nav-2d-utils ros-humble-nav2-behavior-tree ros-humble-nav2-core ros-humble-nav2-costmap-2d ros-humble-phidgets-spatial ros-humble-realsense2-camera-msgs ros-humble-robot-localization ros-humble-rtcm-msgs ros-humble-rviz-imu-plugin ros-humble-rviz-satellite]
#28 27.51 executing command [sudo -H pip3 install -U utm]
#28 27.51 executing command [sudo -H apt-get install -y espeak iputils-ping python3-pygame python3-pykdl python3-shapely python3-sqlalchemy ros-humble-plotjuggler-ros ros-humble-rosbag2-storage-mcap]
#28 27.51 #All required rosdeps installed successfully
#28 DONE 27.7s
```

I understand why `pip3 install` done seperately of course but I wonder why there are still 2 `apt-get` commands instead of one. It's already a huge improvement from:

```
#38 58.68 executing command [sudo -H apt-get install -y iputils-ping]
#38 58.68 executing command [sudo -H apt-get install -y ros-humble-foxglove-bridge]
#38 58.68 executing command [sudo -H apt-get install -y ros-humble-realsense2-camera-msgs]
#38 58.68 executing command [sudo -H apt-get install -y python3-mock]
#38 58.68 executing command [sudo -H apt-get install -y ros-humble-rtcm-msgs]
#38 58.68 executing command [sudo -H apt-get install -y ros-humble-robot-localization]
#38 58.68 executing command [sudo -H apt-get install -y ros-humble-phidgets-spatial]
#38 58.68 executing command [sudo -H pip3 install -U utm]
#38 58.68 executing command [sudo -H apt-get install -y python3-pykdl]
#38 58.68 executing command [sudo -H apt-get install -y ros-humble-rosbag2-storage-mcap]
#38 58.68 executing command [sudo -H apt-get install -y ros-humble-plotjuggler-ros]
#38 58.68 executing command [sudo -H apt-get install -y espeak]
#38 58.68 executing command [sudo -H apt-get install -y ros-humble-foxglove-msgs]
#38 58.68 executing command [sudo -H apt-get install -y python3-pygame]
#38 58.68 executing command [sudo -H apt-get install -y python3-pytest-cov]
#38 58.68 executing command [sudo -H apt-get install -y python3-imageio]
#38 58.68 executing command [sudo -H apt-get install -y ros-humble-nav2-behavior-tree]
#38 58.68 executing command [sudo -H apt-get install -y ros-humble-nav-2d-utils]
#38 58.68 executing command [sudo -H apt-get install -y ros-humble-nav-2d-msgs]
#38 58.68 executing command [sudo -H apt-get install -y ros-humble-nav2-core]
#38 58.68 executing command [sudo -H apt-get install -y ros-humble-nav2-costmap-2d]
#38 58.68 executing command [sudo -H apt-get install -y ros-humble-rviz-imu-plugin]
#38 58.68 executing command [sudo -H apt-get install -y ros-humble-rviz-satellite]
#38 58.68 executing command [sudo -H apt-get install -y python3-sqlalchemy]
#38 58.68 executing command [sudo -H apt-get install -y python3-shapely]
#38 58.68 #All required rosdeps installed successfully
#38 DONE 58.7s
```

SubaruArai, 2024-04-16:
Interesting, gotta dig a bit deeper.
In the original example, `iputils-ping` and `python3-pykdl` were before and after the pip command - yet on the new example, they are on the same line (both after pip).

Do you mind to share on what project structure you ran rosdep on, or even better give me a minimal reproducible repo?

tonynajjar, 2024-04-16:
Here you go
```
<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format3.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="3">
  <name>test_package</name>
  <version>0.0.0</version>
  <description>test_package</description>
  <maintainer email="test">test</maintainer>
  <license>TODO: License declaration</license>

  <depend>rtcm_msgs</depend>
  <depend>robot_localization</depend>
  <depend>phidgets_spatial</depend>
  <depend>python-utm-pip</depend>
  <depend>python3-pykdl</depend>
  <depend>rosbag2_storage_mcap</depend>
  <depend>plotjuggler_ros</depend>
  <depend>utm</depend>

  <test_depend>python3-pytest</test_depend>

  <export>
    <build_type>ament_python</build_type>
  </export>
</package>
```

```
rosdep install --from-paths . --ignore-src -s
#[apt] Installation commands:
  apt-get install ros-humble-phidgets-spatial ros-humble-robot-localization ros-humble-rtcm-msgs
#[pip] Installation commands:
  pip3 install -U utm
#[apt] Installation commands:
  apt-get install python3-pykdl ros-humble-plotjuggler-ros ros-humble-rosbag2-storage-mcap
```

SubaruArai, 2024-04-17:
@tonynajjar There's some discussion on why this slow implementation was used in the pr thread (#953), you might be interested to know the rationale behind.

Ryanf55, 2025-04-03:
You can get a list of keys from rosdep and call apt yourself...  
```
rosdep keys --from-paths src --ignore-src src | xargs rosdep resolve | grep -v \# | xargs sudo apt-get install -y --no-install-suggests --no-install-recommends
```

peci1, 2025-12-01:
If there's a chance more package managers than APT will get involved (like PIP or CARGO in the future), you have to do:

    rosdep keys --from-paths src --ignore-src 2>/dev/null | xargs rosdep resolve 2>/dev/null | grep -v ROSDEP | grep -Pzo '(?s)(?<=\#apt\n)[^#]*' | tr -d '\0' | xargs sudo apt-get install -y --no-install-suggests --no-install-recommends
    rosdep install --from-paths src --ignore src  # install the rest via the other package managers