ros-infrastructure/rosdep #976: Skip all ros packages when using rosdep

xfiderek, 2024-07-06:
Hi!
Is it possible to install only non-ROS related dependencies with rosdep? For example to skip installation of APT packages coming from http://packages.ros.org/ros2/ubuntu repository? Or ideally to throw an error when such packages are to be installed?
The use case that I have is that I am building ROS from source and I want to install non-ROS dependencies such as `python3-sqlalchemy`, but I want to throw an error when packages such as `ros-humble-nav2-msgs` are to be installed via apt.

The approach that I came up with so far is something like:
```bash
set -e
rosdep install --from-paths src -s | grep 'ros-humble' && exit 1
rosdep install --from-paths src
```
I wanted to check if there is a more cleaner way to achieve it.

nuclearsandwich, 2024-08-22:
If you know which ROS packages you are providing, you can skip installing them with the with `--skip-keys`.

If your entire ROS from source configuration is in your `src` directory, you can use `--ignore-src` which will ignore packages that are found in-workspace.

What you might want to consider instead is using https://github.com/ros-infrastructure/rosinstall_generator to generate the source workspace with all ROS dependencies and then you can use that workspace together with `rosdep install --from-paths src --ignore-src` to get only system-level dependencies.

EzraBrooks, 2025-01-14:
See #983 for why `--skip-keys` does not necessarily work