ros-infrastructure/rosdep #991: install package by rosdep key name fails

christianrauch, 2025-01-01:
The documentation on how to `check` and `install` rosdep packages (https://github.com/ros-infrastructure/rosdep/blob/master/doc/overview.rst#installating-rosdeps) seems to be out of date or at least not complete.

At least, I cannot reproduce the commands there.

Example, installing `std_msgs` for `rolling`:
```sh
docker run -it --rm ubuntu:24.04 bash -c "
export DEBIAN_FRONTEND=noninteractive PIP_BREAK_SYSTEM_PACKAGES=1 && \
apt update && apt -y install python3-pip && \
pip3 install rosdep && \
rosdep init && rosdep update && \
rosdep install -s std_msgs --rosdistro=rolling
"
```

This should produce a similar `#[apt] Installation commands:` output as per documentation, but fails with:
```
ERROR: Rosdep cannot find all required resources to answer your query
Missing resource std_msgs
```

The key works with `rosdep resolve std_msgs --rosdistro=rolling` though.

Can you update the documentation to reflect any recent changes on the user-facing behaviour of rosdep? In particular, how to install packages by their rosdep key name.

cottsay, 2025-01-16:
This mechanism worked in ROS 1 using the `ROS_PACKAGE_PATH` environment variable. That variable is not set in ROS 2, so rosdep can't figure out where the `std_msgs` package is. All of the ROS 2 instructions using rosdep make use of the `--from-path` option.

The docs should indeed be updated.

becschwartz, 2025-03-13:
I've been playing with this; I can't get `--from-paths` to work. 

```
I'm in an ubuntu 22.04, I'm using iron. I've got a venv; and I'm using rosdep from the venv 
(rosdep_venv_play) reschwartz@open-source-reschwartz:~/rosdep$ command -v rosdep
/usr/local/google/home/reschwartz/rosdep_venv_play/bin/rosdep
```

I've got a ros2 workspace, where I can `ros2 pkg list` within my `~/ros2_ws/src` and get a long output, which includes urdf for example. 
```
(rosdep_venv_play) reschwartz@open-source-reschwartz:~/ros2_ws/src$ ros2 pkg list | grep urdf
urdf
urdf_parser_plugin
```

When I `rosdep check urdf --from-paths . --rosdistro=iron`, I get 

```
given path 'urdf' does not exist
``` 

Similarly, when I `rosdep check urdf --rosdistro=iron`, I get 
```ERROR: Rosdep cannot find all required resources to answer your query
Missing resource urdf```

christianrauch, 2025-03-14:
> I've been playing with this; I can't get `--from-paths` to work.

What are you trying to do?

> When I `rosdep check urdf --from-paths . --rosdistro=iron`, I get
> 
> ```
> given path 'urdf' does not exist
> ```
> 
> Similarly, when I `rosdep check urdf --rosdistro=iron`, I get
> 
> ```
> ERROR: Rosdep cannot find all required resources to answer your query
> Missing resource urdf
> ```

As mentioned above, `check` and `install` do not work as expected, since this is not supported in ROS 2 any more (or not yet?). That is why I opened this issue. Maybe this can be made to work by setting `ROS_PACKAGE_PATH`?

If you just want the DEB/RPM package name for one or more packages, then `resolve` is the correct verb:
```
$ . /opt/ros/jazzy/setup.bash 
$ rosdep resolve urdf 
#apt
ros-jazzy-urdf
```

Or via `--rosdistro=` if you have no workspace to source. But you have to write your own script around this to actually install those packages.

becschwartz, 2025-04-24:
`ros2 pkg list` makes use of `ament_prefix_path = os.environ.get(AMENT_PREFIX_PATH_ENV_VAR)` 

rosdep is rosdistro-independent so cannot depend on in-distro dependencies.  

There are 2 options that we are considering as a proper fix forward: 
Move the ament resource index to a distro-independent location
Reimplement support 

In the meantime, we can use some hacks. 

// check that `rosdep install rclcpp` doesn’t work → get 
```
ERROR: Rosdep cannot find all required resources to answer your query
Missing resource rclcpp
```
// A hacky workaround is to export ROS_PACKAGE_PATH like we do in ROS1. 
`export ROS_PACKAGE_PATH="${AMENT_PREFIX_PATH}"`
`printenv | grep ROS_PACKAGE_PATH`
// Then you can run rosdep commands. 
`rosdep install rclcpp` 
```
#All required rosdeps installed successfully
```