ros-infrastructure/rosdep #993: Usage of 'rosdep keys' argument is unclear from documentation

firesurfer, 2025-01-24:
I am currently trying to debug why xacro is not picked up as a dependency from a package (aka not installed in the ci pipeline). 

According to the documentation the `rosdep keys` command can be used to list all dependencies of a package. 

> rosdep keys <stacks-and-packages>...
  list the rosdep keys that the packages depend on.

When i now run:

`rosdep keys --from-paths ./src  <my package name>` from the root directory of my local workspace I always get the error:

`given path '<my package name>' does not exist`

The official documentation: http://docs.ros.org/en/independent/api/rosdep/html/commands.html only provides the same information as the `rosdep --help` command. 

Also the guide in the "normal" documentation: https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Rosdep.html does not provide any further information. 

### Question

So the question is:

1. How do list all dependencies of a package
2. Can I also check if these dependencies are installable

firesurfer, 2025-03-27:
A friendly ping

christophebedard, 2025-03-28:
Looks like `rosdep keys` expects a _path_ to a package, not just a package name. Also, from my understanding, I think `--from-paths` effectively limits how deep the search for keys goes:

1. `rosdep keys <path_to_pkg> --from-paths <path_to_pkg>`: returns the package's direct dependencies (i.e., strictly listed in its `package.xml`)
2. `rosdep keys <path_to_pkg> --from-paths <path>`: returns all keys for direct and indirect dependencies, but only searching in the given path
    - For example, if your package (`<path_to_pkg>`) only depends on `pkgA` and that package is in `<path>`, but `pkgA`'s dependencies are not in `<path>`, then you'd only get `pkgA` + `pkgA`'s direct dependencies.

I agree that the documentation/`--help` is not very helpful. This could probably be made clearer: https://github.com/ros-infrastructure/rosdep/blob/2a0510ecff5330de432860b14720ba7672a7ba28/src/rosdep2/main.py#L102-L103.

christophebedard, 2025-03-28:
> 2. `rosdep keys <path_to_pkg> --from-paths <path>`: returns all keys for direct and indirect dependencies, but only searching in the given path
>     - For example, if your package (`<path_to_pkg>`) [...]

Ah, this might not even be true. The package I tried this on has no dependencies on any packages in `<path>`, and yet I get a bunch of keys, including packages that actually depend _on_ my package.

christophebedard, 2025-03-28:
In any case, if you can figure out how it's meant to work, improving the documentation would definitely be appreciated 😄