ros-infrastructure/rosdep #994: Some packages cannot be found

EmmanuelMess, 2025-02-03:
I am running rosdep install inside a docker build and I am getting missing packages. I've tried reinstalling rosdep, running apt update, to no avail. The packages should exist according to the docs.

Full log:

```
1.072 Warning: running 'rosdep update' as root is not recommended.
1.072   You should run 'sudo rosdep fix-permissions' and invoke 'rosdep update' again without sudo.
9.376 reading in sources list data from /etc/ros/rosdep/sources.list.d
9.376 Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
9.376 Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
9.376 Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
9.376 Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
9.376 Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
9.376 Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
9.376 Skip end-of-life distro "ardent"
9.376 Skip end-of-life distro "bouncy"
9.376 Skip end-of-life distro "crystal"
9.376 Skip end-of-life distro "dashing"
9.376 Skip end-of-life distro "eloquent"
9.376 Skip end-of-life distro "foxy"
9.376 Skip end-of-life distro "galactic"
9.376 Skip end-of-life distro "groovy"
9.376 Add distro "humble"
9.376 Skip end-of-life distro "hydro"
9.376 Skip end-of-life distro "indigo"
9.376 Skip end-of-life distro "iron"
9.376 Skip end-of-life distro "jade"
9.376 Add distro "jazzy"
9.376 Skip end-of-life distro "kinetic"
9.376 Skip end-of-life distro "lunar"
9.376 Skip end-of-life distro "melodic"
9.376 Add distro "noetic"
9.376 Add distro "rolling"
9.376 updated cache in /root/.ros/rosdep/sources.cache
9.901 Reading package lists...
9.920 Building dependency tree...
9.923 Reading state information...
9.926 E: Unable to locate package ros-jazzy-ros2-control
9.934 Reading package lists...
9.951 Building dependency tree...
9.954 Reading state information...
9.958 E: Unable to locate package ros-jazzy-ros2-controllers
9.966 Reading package lists...
9.983 Building dependency tree...
9.986 Reading state information...E: Unable to locate package ros-jazzy-gz-ros2-control
9.989 
9.997 Reading package lists...
10.02 Building dependency tree...
10.02 Reading state information...
10.02 E: Unable to locate package ros-jazzy-urdf-launch
10.18 ERROR: the following rosdeps failed to install
10.18   apt: command [apt-get install -y ros-jazzy-ros2-control] failed
10.18   apt: command [apt-get install -y ros-jazzy-ros2-controllers] failed
10.18   apt: command [apt-get install -y ros-jazzy-gz-ros2-control] failed
10.18   apt: command [apt-get install -y ros-jazzy-urdf-launch] failed
10.18   apt: Failed to detect successful installation of [ros-jazzy-ros2-control]
10.18   apt: Failed to detect successful installation of [ros-jazzy-ros2-controllers]
10.18   apt: Failed to detect successful installation of [ros-jazzy-gz-ros2-control]
10.18   apt: Failed to detect successful installation of [ros-jazzy-urdf-launch]
10.18 executing command [apt-get install -y ros-jazzy-ros2-control]
10.18 executing command [apt-get install -y ros-jazzy-ros2-controllers]
10.18 executing command [apt-get install -y ros-jazzy-gz-ros2-control]
10.18 executing command [apt-get install -y ros-jazzy-urdf-launch]

```

The commands:

```
#!/usr/bin/env bash                                                            \n\
                                                                               \n\
set -o errexit      # Exit on most errors (see the manual)                     \n\
# TODO set -o nounset      # Disallow expansion of unset variables             \n\
set -o pipefail     # Use last non-zero exit code in a pipeline                \n\
                                                                               \n\
# Enable errtrace or the error trap handler will not work as expected          \n\
set -o errtrace         # Ensure the error trap handler is inherited           \n\
                                                                               \n\
source /opt/ros/${ROS_DISTRO}/setup.bash                                       \n\
rosdep update                                                                  \n\
rosdep install --from-paths ./src --ignore-src --rosdistro ${ROS_DISTRO} -r -y \n\
rosdep check --from-paths ./src --ignore-src                                   \n\
rm -rf /var/lib/apt/lists/*                                                    \n\
colcon build                                                                   \n\

```

EmmanuelMess, 2025-02-03:
Okay, apparently `rosdep update` doesn't run `apt-get update` which makes no sense?

christophebedard, 2025-02-04:
It makes sense if you consider what `rosdep update` is _supposed_ to do 😄

[`rosdep` has its own list/index](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Rosdep.html#how-does-rosdep-work) that it needs to retrieve. That's [what `rosdep update` does](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Rosdep.html#rosdep-operation). So it does mean that you have to run `apt update` beforehand. This same topic was discussed in #904.

tfoote, 2025-02-04:
rosdep update also doesn't manage your sources.list files either. Generally you should be running apt-get update after you install a new sources.list. The apt dependencies also may go out of date at any time. And you hit this commonly in docker images because you want to keep the size down so the cache is removed at every step. I see that you've included the cleanup of `/var/lib/apt/lists/*` in your script so you're familiar with that process. And as such you should be restoring that cache before you try to use it.

I'll close this as duplicate of #904 but even with that still open, I think that it's likely that expanding the scope of rosdep to include this is not a good idea. I'd be more in favor of adding a console output in case of failure suggesting common failures such as not having the right sources.list.d files or not having run apt-get update.