ros-infrastructure/rosdep #1005: Allow rosdep install -t exec on install directory

Ryanf55, 2025-05-28:
Use Case:

I have compiled a package using Qemu on AMD64 for an ARM64 target. Then, I copy the `install` directory of the package to the target (Raspberry Pi).
This package has runtime dependencies specified in rosdep.

I would like to be able to invoke `rosdep install --from-paths install -t exec` to get all the binary runtime dependencies installed.

# Expected behavior

Rosdep will install runtime dependencies for all packages in the install folder

# Actual Behavior

Rosdep doesn't see any packages in the install folder. This makes dockerless deployment hard because now a user has to clone the source on target hardware to install dependencies

# Alternatives considered

Use the oneshot list generated on amd64 from #953 and run the same command on the target (assumes package names are the same for amd64 and arm64).

christophebedard, 2025-05-28:
Can you try deleting the `COLCON_IGNORE` file under `install/` and trying again? I think `rosdep` intentionally ignores anything under `install/`.

christophebedard, 2025-05-28:
First of all, my comment above wasn't really correct: `rosdep` isn't the one that is intentionally ignoring `install/`. That's done by [`catkin_pkg`](https://github.com/ros-infrastructure/catkin_pkg/blob/eec17e72cbe3c25173ec0cc8868106904397c560/src/catkin_pkg/packages.py#L43), which crawls directories and finds packages. However, by default, `rosdep` should of course ignore `install/`, otherwise you get duplicate packages when running `rosdep` from the workspace root. It's just meant to run against the packages in the workspace.

All of this is to say that I think this belongs in a more specific guide/tutorial (e.g., cross-compilation) rather than `rosdep` itself. I don't want to prevent adding useful documentation, but I'm not sure it belongs here directly.

But I'd like to hear what @cottsay has to say.