ros-infrastructure/rosdep #1008: Failure building package from source in container CI

VintageDoggo, 2025-06-23:
## Issue
When building a package from source inside an OCI container inside GitLab's CI, `rosdep` returns an internal error. 
A local build succeeds when building the package using docker and inside a devcontainer. 
```
ERROR: Rosdep experienced an internal error.
Please go to the rosdep page [1] and file a bug report with the message below.
[1] : http://www.ros.org/wiki/rosdep
rosdep version: 0.25.1
Bad installer [dnf]: [Errno 2] No such file or directory: 'rpm'
subprocess exited with status 1
subprocess exited with status 1
Error: building at STEP "RUN rosdep install --from-paths src --ignore-src --rosdistro ${ROS_DISTRO} -y     && colcon build --event-handlers console_direct+ --cmake-args -DCMAKE_BUILD_TYPE=Release": exit status 1
```
## Expected behaviour 
Package built

## Actual behaviour
Returns `rosdep` internal error 

## Details 

- Distro: Jazzy
- Podman: v5.4.0
- Package: [rmw_zenoh](https://github.com/ros2/rmw_zenoh) 

### How to reproduce

**Dockerfile extract:**
```
ARG ROS_DISTRO="jazzy"
ARG ARCH="amd64"
# ********************************************************
# ***************** BASE + UNDERLAY IMAGE ****************
# ********************************************************
VOLUME "/var/lib/containers"
FROM --platform=linux/${ARCH} docker.io/ros:${ROS_DISTRO} AS underlay
ENV ROS_DISTRO=${ROS_DISTRO}

#ENV ROS_AUTOMATIC_DISCOVERY_RANGE=LOCALHOST
SHELL ["/bin/bash", "-c"]

RUN mkdir -p /ros2_ws/src
WORKDIR /ros2_ws/src

# Pull any of the needed dependencies
# Add rmw_zenoh from source to build 0.2.5
RUN git clone https://github.com/ros2/rmw_zenoh.git -b jazzy

# Setup the workspace
WORKDIR /ros2_ws
RUN source /opt/ros/${ROS_DISTRO}/setup.bash \
    && apt-get update && apt-get upgrade -y \
    && apt-get install -y python3-pip libusb-1.0.0-dev \ 
    && rosdep install --from-paths src --ignore-src --rosdistro ${ROS_DISTRO} -y \
    && colcon build --event-handlers console_direct+ --cmake-args -DCMAKE_BUILD_TYPE=Release

ENV RMW_IMPLEMENTATION=rmw_zenoh_cpp
```
**Podman commands:**
`podman login mygitlabregistry.com`

```podman build --build-arg ARCH=arm64 -f "${CI_PROJECT_DIR}/docker/dockerfile-ros-base" --format docker --platform=$ARM_PLAT -t $CI_REGISTRY/engineering/ros-docker:base-${arch} .```

```podman push $CI_REGISTRY/my/registry/ros-docker:base-${arch}```

**Docker commands for local build:**

```docker build --build-arg ARCH=arm64 -f "docker/dockerfile-ros-base" --platform=$ARM_PLAT -t $CI_REGISTRY/engineering/ros-docker:base-${arch} .```

christophebedard, 2025-06-23:
> ```
> Bad installer [dnf]: [Errno 2] No such file or directory: 'rpm'
> ```

You're expecting the base image to be Ubuntu, right? It seems like `rosdep` detects RHEL/Fedora/etc. and tries to use `dnf`. Without being able to reproduce it, it's hard to say. Some things to try:

1. For debugging:
    1. Add `python3 -m rospkg.os_detect` to the commands being run before `rosdep install`. What's the output? On my Ubuntu 24.04 system:
        ```console
        $ python3 -m rospkg.os_detect
        OS Name:     ubuntu
        OS Version:  24.04
        OS Codename: noble
        ```
    1. Print the content of `/etc/os-release`. That's what `rospkg` uses to detect the OS: https://github.com/ros-infrastructure/rospkg/blob/db7614e5209137faa6ec01e2edaf34f775780b1a/src/rospkg/os_detect.py#L196-L204.
1. Add `--os=ubuntu:noble` to `rosdep install` to override the OS detection mechanism.

VintageDoggo, 2025-06-24:
Hi!
Thank you for the response, yes the base image is to be Ubuntu Noble, so overriding the os with `--os=ubuntu:noble` as you pointed out did the trick, since I suppose it was detecting the container image I was using to build the image (which was indeed RHEL). 
Thank you!

closed 2025-06-24 by VintageDoggo

VintageDoggo, 2025-06-24:
Sure! 

Output of `python3 -m rospkg.os_detect`:
```
[2/3] STEP 8/11: RUN python3 -m rospkg.os_detect
OS Name:     fedora
OS Version:  41
OS Codename: container image
```

Contents of `/etc/os-release`:
```
NAME="Fedora Linux"
VERSION="41 (Container Image)"
RELEASE_TYPE=stable
ID=fedora
VERSION_ID=41
VERSION_CODENAME=""
PLATFORM_ID="platform:f41"
PRETTY_NAME="Fedora Linux 41 (Container Image)"
ANSI_COLOR="0;38;2;60;110;180"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:41"
DEFAULT_HOSTNAME="fedora"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f41/system-administrators-guide/"
SUPPORT_URL="https://ask.fedoraproject.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=41
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=41
SUPPORT_END=2025-12-15
VARIANT="Container Image"
VARIANT_ID=container
```

christophebedard, 2025-06-24:
Thanks! That's definitely weird.

I suspect that there might be something wrong with the way Docker containers are configured/run on your podman or GitLab CI infrastructure. If I just run a Fedora 41 container on my Ubuntu 24.04 host (other way around compared to you), I correctly get the Fedora 41 info from `/etc/os-release` inside the container:

```
$ docker run -it fedora:41
[root@7fc0405f7746 /]# cat /etc/os-release 
NAME="Fedora Linux"
VERSION="41 (Container Image)"
RELEASE_TYPE=stable
ID=fedora
VERSION_ID=41
VERSION_CODENAME=""
PLATFORM_ID="platform:f41"
PRETTY_NAME="Fedora Linux 41 (Container Image)"
ANSI_COLOR="0;38;2;60;110;180"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:41"
DEFAULT_HOSTNAME="fedora"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f41/"
SUPPORT_URL="https://ask.fedoraproject.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=41
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=41
SUPPORT_END=2025-12-15
VARIANT="Container Image"
VARIANT_ID=container
```

However, if I somehow mount `/etc/os-release` from my host into the container, then `/etc/os-release` inside the container reflects the host:

```
$ docker run -it -v /etc/os-release:/etc/os-release fedora:41
[root@610a47a52ffc /]# cat /etc/os-release 
PRETTY_NAME="Ubuntu 24.04.2 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04.2 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo
```

Anyway, if you're fine with adding `--os=ubuntu:noble`, all good. Otherwise, this might be worth looking into, since it could lead to other problems eventually if this is really the case.

VintageDoggo, 2025-06-25:
That is strange. If I have some time, I'll look into this and post any discoveries I can make. For now overriding the OS works like a charm. 
Thank you for checking this further!